NXPNFCLinux/linux_libnfc-nci #84: Can't seem to run the demo app

courtneyconroy, 2018-07-25:
Does anyone have a bit more detail on setting this up? I just cloned the github but when I try ./nfcDemoapp it doesn't work. Is there something I am missing? This is my first time doing such a thing

jgeslin, 2018-07-25:
Details are given in related documentation: https://www.nxp.com/docs/en/application-note/AN11697.pdf.
Which target are you using ? Raspberry Pi ?

courtneyconroy, 2018-07-26:
We are using Raspberry Pi but we can't find a dts file to continue on with the steps

jgeslin, 2018-07-26:
If you are using Raspberry Pi, you can use the "alternative to kernel driver" configuration. After cloning the repo, and enabling the i2c-dev driver in kernet (through raspi-config command) just do (from the linux_libnfc-nci directory):
`./bootstrap`
`./configure --enable-pn7150 --enable-alt --sysconfdir=/etc` // if using OM5578, otherwise if using OM5577 demo kit define "--enable-pn7120" instead of "--enable-pn7150
`make`
`sudo make install`
`./nfcDemoApp poll`

closed 2018-11-07 by jgeslin