NXPNFCLinux/linux_libnfc-nci #85: how to read language_code?

jjvbsag, 2018-09-26:
in function
```c
int ndef_createText(
  char *language_code, 
  char *text,
  unsigned char *out_ndef_buff, 
  unsigned int out_ndef_buff_length
)
```
there is a parameter `language_code` to be written into a tag.

**But**
When reading a tag, 
```c
int ndef_readText(
  unsigned char *ndef_buff, 
  unsigned int ndef_buff_length,
  char * out_text, 
  unsigned int out_text_length
)
```
only provides an interface to read back `text`, but not `language_code`

By which interface can I read back  `language_code` ?

Should readText not be
```c
int ndef_readText(
  unsigned char *ndef_buff, 
  unsigned int ndef_buff_length,
  char * out_language_code, 
  unsigned int out_language_code_length
  char * out_text, 
  unsigned int out_text_length
)
```
???

jjvbsag, 2018-09-26:
or add an
```c
int ndef_readLanguageText(
  unsigned char *ndef_buff, 
  unsigned int ndef_buff_length,
  char * out_language_code, 
  unsigned int out_language_code_length
  char * out_text, 
  unsigned int out_text_length
)
```
if you want to keep the API compatible

jgeslin, 2019-04-01:
See commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/67022903e6b45e084ba7129675c3f4fdff8ba847.

New API added to allow retrieving the language code:
```
int ndef_readLanguageCode(
  unsigned char *ndef_buff, 
  unsigned int ndef_buff_length,
  char * out_language_code, 
  unsigned int out_language_code_length
)
```
It was not possible to combine both reading language code and text message in the same API because the function must return the size of the data read and thus cannot return both sizes (text message size and language code size).

jjvbsag, 2019-07-15:
Sorry, for the delay.
I just verified the master and `ndef_readLanguageCode` is doing fine. Thanks.
If this is ok for you, I'd leave this issue open until the next Release of `linux_libnfc-nci`, verify the release and then close the issue

jgeslin, 2019-09-06:
Closing since release in R2.4.1

closed 2019-09-06 by jgeslin