NXPNFCLinux/linux_libnfc-nci #91: PN7150 stopped working permanently

orange-kao, 2018-12-06:
Hello.

We have been using our boards (battery powered) with PN7150 for a while and there are few (3+) cases that PN7150 stopped working permanently. Power cycle, reflash OS or reinstall libnfc-nci does not recover PN7150 from this issue. The code is derived from commit a085cc8d09c3d0b1564d0c3883cf3f65eec806c0.

When we run "nfcDemoApp poll", message "Waiting for a Tag/Device..." does not show up on affected boards.

Debug message indicates `Max retry count for RF config FW recovery exceeded`. Is it the same issue as issue #59?

Full log as attached. Thank you.
```

#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

1544057441:663:445 - NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0xff)

1544057441:663:573 - NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0xff)

1544057441:663:607 - NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0xff)

1544057441:663:636 - NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0xff)

0:000:270 - NxpHal:     Entering phNxpNciHal_init_monitor
0:000:322 - NxpHal:     Returning with SUCCESS
0:000:397 - NxpExtns:   find found CUST_PIN_VEN_INVERT=(0x0)

0:000:432 - NxpExtns:   find found CUST_I2C_BUS=(0x1)

0:000:467 - NxpExtns:   find found CUST_I2C_SLAVE=(0x28)

0:000:497 - NxpExtns:   find found CUST_PIN_IRQ=(0x80)

0:000:527 - NxpExtns:   find found CUST_PIN_VEN=(0x8a)

0:000:554 - NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

0:000:578 - NxpTml:     NFC - Assign IO pins

0:000:609 - NxpTml:     Pin /sys/class/gpio/gpio128

0:000:785 - NxpTml:     System already has pin /sys/class/gpio/gpio128

0:000:827 - NxpTml:     Unexport pin 128

0:001:219 - NxpTml:     Export pin 128

0:001:617 - NxpTml:     Direction /sys/class/gpio/gpio128/direction

0:001:714 - NxpTml:     Direction port '/sys/class/gpio/gpio128/direction' ready
0:001:767 - NxpTml:     Pin 128 now an input

0:001:804 - NxpTml:     Edge /sys/class/gpio/gpio128/edge

0:001:880 - NxpTml:     Edge-string rising - 6

0:001:953 - NxpTml:     Pin 128 trigger on rising

0:001:990 - NxpTml:     Value /sys/class/gpio/gpio128/value

0:002:075 - NxpTml:     Pin /sys/class/gpio/gpio138

0:002:151 - NxpTml:     System already has pin /sys/class/gpio/gpio138

0:002:183 - NxpTml:     Unexport pin 138

0:002:387 - NxpTml:     Export pin 138
[... 1389 lines cut ...]
2:960:152 - NxpNciX:     len = 166 > 2002A313A00D0604359001F401A00D06064401900300A00D060630B0011000A00D0606420200FFFFA00D03063F04A00D0620428800FFFFA00D0422442200A00D06222D50340C00A00D063242F800FFFFA00D06342D24370C00A00D06343386800070A00D0434442200A00D06422D15450D00A00D0446442200A00D06462D05590E00A00D0644428800FFFFA00D06562D059F0C00A00D0654428800FFFFA00D060A3380860070
2:960:238 - NxpTml:     PN54X - I2C Write successful.....

2:960:266 - NxpTml:     PN54X - Posting Fresh Write message.....

2:960:308 - NxpTml:     PN54X - Tml Writer Thread Running................

2:960:353 - NxpHal:     write successful status = 0x0
2:960:415 - NxpHal:     Response timer started
2:960:445 - NxpHal:     Waiting after ext cmd sent
2:998:062 - NxpTml:     PN54X - I2C Read successful.....

2:998:142 - NxpNciR:     len =   9 > 4002060902A00DA00D
2:998:170 - NxpTml:     PN54X - Posting read message.....

2:998:228 - NxpHal:     read successful status = 0x0
2:998:265 - NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_INVALID_PARAM
2:998:321 - NxpHal:     Response timer stopped
2:998:351 - NxpHal:     Checking response
2:998:379 - NxpHal:     Max retry count for RF config FW recovery exceeded 
2:998:473 - NxpTml:     PN54X - Read requested.....

2:998:505 - NxpTml:     PN54X - Invoking I2C Read.....

```

jgeslin, 2018-12-06:
Hi,

It sounds like a EEPROM memory corruption issue as depicted in PN7150 datasheet (https://www.nxp.com/docs/en/data-sheet/PN7150.pdf) chapter 11 (see "Important" statement).
Previous execution of RF settings write operation should have been interrupted by power-off or HW reset.

Unfortunately there is no way to recover those IC.

To prevent this issue in future, please make sure you are using latest linux_libnfc stack (including commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/edd10dccc34e168469820a021b0f0cc480c1ce68) which avoid EEPROM write operation while no changes are observed (so RF settings should only be applied once).

Regards,
Jeremy.

Regards,
Jeremy.

sschueller, 2022-06-16:
I get this in my Montblanc summit 2 smart watch logs running Google's WearOS. NFC stopped working all of sudden after 2 years, reset of watch has no effect. NFC does not work anymore. 

So from what I can read here it appears that either this was improperly implemented on my watch (WearOS?) or some condition occurred where this could happen. Either way it seems like a bad idea that such a thing can occur on a watch where the chip can't easily be replaced.

As my watch is out of warranty it will not be replaced and Montblanc does not offer repair for a device that they no longer sell.

So my questions is, if I replace (very difficult from what it appears: https://fccid.io/2AENP-S2C19/Internal-Photos/Internal-photos-4509048) the chip which seems to be impossible to purchase (NXP Q3304, anyone have a source of even a datasheet?) so I have to find a donor, would it work if the IC isn't also corrupted on the donor?