NXPNFCLinux/linux_libnfc-nci #93: Disable CRC in nfcTag_transceive()

orange-kao, 2018-12-07:
Hello.

I am working with Mifare Classic and I can authenticate by sending the key A or B over the air in plaintext via nfcTag_transceive(). 

If I want to use mutual 3-pass authentication on Mifare Classic (chapter 12.1, [Mifare Classic EV1 1K  datasheet MF1S50YYX_V1](https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf)), I may need to disable CRC. Does there any way to disable CRC in nfcTag_transceive()?

Thank you.

jgeslin, 2018-12-07:
Hi,

As shown in https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/tree/master/tag-raw_example, MIFARE mutual 3-pass authentication procedure is handled by the NFC Controller.
For the specific case of such commands, the libnfc-nci stack drives the NFC controller to run the Authentication through only one nfcTag_transceive() call. The input parameter is then defined as is:
 - 0x60 or 0x61 (defining authenticate cmd) 
 - Address (1 byte defining MIFARE Block address)
 - UID (4 bytes card UID ... can be let to 0x00 since the stack will anyway handle it for activated card)
 - Key (6 bytes key used for authentication)

Regards,
Jeremy.

jgeslin, 2019-04-01:
Considered answered.

closed 2019-04-01 by jgeslin