NXPNFCLinux/linux_libnfc-nci #95: How to use library to program NTAG413 DNA tags?

shaunyu, 2019-03-25:
I am trying to use this library to program the NTAG413 DNA tags. My approach is to use this library with the Beaglebone Black kit and run as such:
i) Detect tag arrival
ii) Write NDEF file
ii) AuthenticateFirst with Key0
iii) Write CC file

Any ideas how to begin?

Rgds,
Shaun

jgeslin, 2019-03-25:
Hi,

From what I know about NTAG413 DNA, it is behaving as NFC Forum T4T and implement additionally ISO7816 proprietary APDU for supplementary fonctions.

In https://github.com/NXPNFCLinux/linux_libnfc-nci_examples you'll find simple code example for NDEF tag writing (https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/tree/master/tag-write_example) and raw tag access (https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/tree/master/tag-raw_example).

Combining both examples you should be able to implement this scenario. See attached a draft version using "nfcTag_transceive" API of the library to send Cmd.GetVersion command.
[tag-write_example.tar.gz](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3004558/tag-write_example.tar.gz)

Regards,
Jeremy.

shaunyu, 2019-03-26:
Thanks for the example. I will try it out.  Is there anything to append to those APDUs defined in AN11910 or just send them in using the nfcTag_transceive() method will do?

jgeslin, 2019-03-26:
The "nfcTag_transceive" API allows exchanging APDU with ISO14443-4 tag in raw (nothing to be added or removed).

shaunyu, 2019-03-26:
Good to know. Thanks for the prompt reply. By the way, is there any documentation as to what those response from GetVersion C-APDU means? Can't find any documentation that states how to interpret the bytes.

jgeslin, 2019-03-26:
I would advise you to ask this question about NTAG413 DNA on https://community.nxp.com/.

shaunyu, 2019-03-26:
I tried to send the C-APDU to select NDEF Application and it returns 0x69, 0x85 which says "No currently selected EF, no command to monitor / no Transaction Manager File" according to a website I found. My C-APDU is {0x00, 0xA4, 0x04, 0x00, 0x07, 0xD2, 0x76, 0x00, 0x00, 0x85, 0x01, 0x01, 0x00}.  Below is the snippet I added to your tag-write_example:

/*
  * Select NDEF 
  */
  memset(Resp, 0, 16);
  res = nfcTag_transceive(g_tagInfos.handle, SelectNDEFApplication, sizeof(SelectNDEFApplication), Resp, 16, 1000);
  printf("res=%d\n", res);
  if (res > 0) 
      printHex("Select NDEF App", Resp, res);

Any idea?

jgeslin, 2019-03-26:
From my mind NTAG413 DNA comes already formatted (so NDEF application should be present).
Did you tried the nfcTag_isNdef ?
Or can you run the demo application "nfcDemoApp poll" and tap your tag ?

shaunyu, 2019-03-26:
I am using a tag that I have programmed using the NXP Pegoda reader and a demo tool. And using the tag_read-example, I can read the NDEF in it.  But using your tag-write_example, when I call the nfcTag_isNDEF(), it returns 0. Below is the snippet:

    printf("\n-------------\nWaiting for tag...\n");
    do{
        pthread_cond_wait(&condition, &mutex);

        /* Check if discovered tag is ISO14443-4 */ 
        if(g_tagInfos.protocol == NFA_PROTOCOL_ISO_DEP) 
        {
            /* Check if discovered tag is NTAG413 DNA */ 
            res = nfcTag_transceive(g_tagInfos.handle, GetVersion, sizeof(GetVersion), Resp, 16, 500);
            if(0x00 == res) {
                printf("Get Version failed\n");
            }
            else {

                /*
                 * TBD: check if GetVersion response indicates NTAG413 DNA 
                 */
                  if (isNTag413DNA(NTag413DNAVersion, Resp, res)) {
                            printf("Tag is 413DNA\n");
                            
                             res = nfcTag_isNdef(g_tagInfos.handle, &NDEFinfo); 
                             printf("res=%d\n", res); //<----- here res = 0
                             if(0x01 == res) {
                                    NDEFContent = malloc(NDEFinfo.current_ndef_length * sizeof(unsigned char));
                                    res = nfcTag_readNdef(g_tagInfos.handle, NDEFContent, NDEFinfo.current_ndef_length, &lNDEFType);
                             }
                  }

I can try other 413DNA samples and revert. Really appreciate your help here. Else will be roaming in the dark.

jgeslin, 2019-03-26:
I don't have NTAG413 DNA so I implemented similar scenario using MIFARE DESFire (also behaving as T4T and supporting proprietary APDU GetVersion command).
See my example attached.
[tag-write_example.tar.gz](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3008972/tag-write_example.tar.gz)

shaunyu, 2019-03-27:
I will try and revert. Thanks for the help :).

shaunyu, 2019-03-27:
It is now selecting NDEF Application and NDEF File correctly. Not sure why it didn''t work the first time.  Thanks for all the help :)
By the way, I notice that there is a difference in documentation about the C-APDU for Select.CMD for NDEF application. According to the AN11910, the example shows the P1=0x04, P2=0x0C wheres in the example code P2=0x00.  Inside the NT4H1321, table 7, it shows P1=P2=XX. Where can I find more information in these?

jgeslin, 2019-03-27:
As said before:
> I would advise you to ask this question about NTAG413 DNA on https://community.nxp.com/.

shaunyu, 2019-03-28:
Using your example with nfcTag_transceive(), I can not select NDEF application, NDEF file and write NDEF File. However, Authenticate.First returned 0. In fact, I tried another command Read NDEF and it returned 0 also. Under what condition will it return 0? No status words so cannot check what is wrong.

// based on AN11910 (ver1.5)
#define LE 0x00
unsigned char ReadNDEFFile[] = {0x90, 0xAD, 0x00, 0x00, 0x07, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, LE};
unsigned char AuthenticateFirst1[] = {0x90, 0x71, 0x00, 0x00, 0x02, 0x00, 0x00, LE};

Appreciate your help in this matter.
Thanks.

jgeslin, 2019-03-28:
What do you mean by:
> Authenticate.First returned 0

Can you share NCI logs ?

shaunyu, 2019-03-28:
res = nfcTag_transceive(g_tagInfos.handle, AuthenticateFirst1, sizeof(AuthenticateFirst1), Resp, 16, 1000);
printf(res); // res = 0;

shaunyu, 2019-03-29:
I have attached the log for NCIX_LOGLEVEL=0x03, NCIR_LOGLEVEL=0x03, NCIHAL_LOGLEVEL=0x03. I have selected the NDEF application, NDEF File and then AuthenticateFirst.  Thanks.

[AuthenticateFirstLog.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3020606/AuthenticateFirstLog.txt)

shaunyu, 2019-03-29:
From the logs, it seems like there is an AF returned. But the byte count (I assumed) is 0x12 which is wrong?
Authenticate First Part 1
NxpNciX:     len =  11 > 0000089071000002000000
NxpHal:     write successful status = 0x0
NxpNciR:     len =   6 > 600603010001
NxpHal:     read successful status = 0x0
NxpNciR:     len =  21 > 000012F4549DE2FDC5994EA30DB995920531F6**91AF**
NxpHal:     read successful status = 0x0

=AUTH 1[0 bytes]==== // the 0 bytes is the printf of res = nfcTag_transceive(...)

jgeslin, 2019-03-29:
Indeed, AuthenticateFirst command is a two pass command as shown in below picture.

[image]

shaunyu, 2019-03-29:
But shouldn't nfcTag_transceive() returns 21 instead of zero?

shaunyu, 2019-03-29:
Jeremy, let me know if you needed more logs. Seems like a bug in the nfcTag_transceiver() function. How to enable the API_LOG to view what is happening inside the transceive function?

Thanks.

jgeslin, 2019-03-29:
nfcTag_transceive() returns 0 because you indicated 16 as reponse buffer size in the parameters.
You should rather use:
`res = nfcTag_transceive(g_tagInfos.handle, AuthenticateFirst1, sizeof(AuthenticateFirst1), Resp, sizeof(Resp), 1000);
`

shaunyu, 2019-03-30:
My bad. It's working now. Thanks for the help.

jgeslin, 2019-04-01:
Considered answered.

closed 2019-04-01 by jgeslin