NXPNFCLinux/linux_libnfc-nci #96: nfcDemoApp : i2c_master_send returns negative value

jackofheartss, 2019-03-25:
Hello,

I followed this guide : https://www.nxp.com/docs/en/application-note/AN11697.pdf to install the NXP driver and NFC library onto an embedded Linux system : 

```
u-name -r
4.9.88-10180-gaaecf0961617-dirty
```
I modified the dtsi file as follows : 

```
&i2c1 {
        clock-frequency = <100000>;
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_i2c1>;
        status = "okay";

        pn547: pn547@28 {
        compatible = "nxp,pn547";
        reg = <0x28>;
        clock-frequency = <400000>;
        interrupt-parent = <&gpio2>;
        interrupt-gpios = <&gpio2 15 0>;
        enable-gpios = <&gpio2 8 0>;
        };
};

                pinctrl_i2c1: i2c1grp {
                        fsl,pins = <
                                MX6UL_PAD_GPIO1_IO02__I2C1_SCL          0x4001b8b0
                                MX6UL_PAD_UART4_RX_DATA__I2C1_SDA       0x4001b8b0
                        >;
                };

```

But when I run nfcDemoApp, I never have the chance to see "Waiting for a Tag/Device", instead I get the following error :

```
################################################################################                                                                                                                               #########
##                                       NFC demo                                                                                                                                                                     ##
################################################################################                                                                                                                               #########
##                                 Poll mode activated                                                                                                                                                                ##
################################################################################                                                                                                                               #########
                              ... press enter to quit ...

pn54x_dev_open : 10,55
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=0
pn544_disable power off
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_write : i2c_master_send returned -6
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=0
pn544_disable power off
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=0
pn544_disable power off
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=0
pn544_disable power off
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
```

What could be the reasons for such an issue ? 
It does not seem that I have problems on my i2c bus as I am able to see (i2cdetect), write (i2cset) and read (i2cget) what's on my NFC tag reader. 

Thank you in advance for your ideas

jgeslin, 2019-03-25:
Hi,

From the log it seems NFC device (NFC controller IC) is not answering to the I2C request (not acknowledging the address). 
First check your HW phyisical connections between you platform and the NFC device (I2C lines, VEN and IRQ).
Which NFC Controller are you using ? 

Regards,
Jeremy.

jackofheartss, 2019-03-26:
I checked my hardware, everything looks fine.
I used this NFC controller : https://www.nxp.com/docs/en/data-sheet/PN7150.pdf

jackofheartss, 2019-03-26:
I enabled the debug in /etc/lib-nxp-init.conf to 0x03 and get this :

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure

NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpHal:     thread started
NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb -pn54x_dev_ioctl, cmd=1074063617, arg=1
 write timeout!!!
NxpHal:     Repn544_enable power on
sponse timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpHal:     PN54X Reset - SUCCESS

NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=0
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb -pn54x_dev_ioctl, cmd=1074063617, arg=1
 write timeout!!!
NxpHal:     Repn544_enable power on
sponse timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpHal:     PN54X Reset - SUCCESS

NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=0
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb -pn54x_dev_ioctl, cmd=1074063617, arg=1
 write timeout!!!
NxpHal:     Repn544_enable power on
sponse timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpHal:     Power Cycle
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=1
phTmlNfc_i2c_reset(), VEN level 1pn544_enable power on

NxpHal:     PN54X Reset - SUCCESS

NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=0
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpHal:     hal_extns_write_rsp_timeout_cb - write timeout!!!
NxpHal:     Response timer stopped
NxpHal:     Callback Status is failed!! Timer Expired!! Couldn't read it! 0xff
NxpHal:     NCI_CORE_RESET: Failed
NxpTml:     PN54X - Write request NOT enabled

jgeslin, 2019-03-26:
That's make sense:
The first log shows PN7150 acknowledging the I2C address and receiving the first command (NCI CORE_RESET_CMD "20000101"). The IRQ line going high is signaling PN7150 wants to answer. The NFC driver will only perform I2C read request (to read PN7150 answer) when detecting IRQ line going high. So you must fix the erroneous GPIO status to have it working.

Regarding the second log and your comment: "But when I run the command for the very first time, I have two more lines :". This is explained by the PN7150 standby mode. When in this mode, PN7150 doesn't ack the first I2C request, and the NFC driver is retrying.

jackofheartss, 2019-04-02:
Thank you for your help, it turns out I was having conflicts in my dtsi files. To make it short : the IRQ pin was being used by on of my Ethernet port so I changed the Ethernet port. 

Now when I run nfcDemoApp I can see this : 

```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                          pn54x_dev_open : 10,55
    ... press enter to quit ...
pn54x_dev_ioctl, cmd=1074063617, arg=1

pn544_enable power on
pn54x_dev_ioctl, cmd=1074063617, arg=0
pn544_disable power off
pn54x_dev_ioctl, cmd=1074063617, arg=1
pn544_enable power on
Waiting for a Tag/Device...
```

Which is what I expected. But whenever I put an NFC tag near the antenna, nothing happens and nothing shows up on my terminal. 
What could be the reasons for such a behavior ? Is there an expected behavior for IRQ or VEN pins when a tag is near the antenna ? 

Here is the full log when lauching nfcDemoApp poll :

```
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
pn54x_dev_open : 10,55        ... press enter to quit ...

NxpExtns:   find found NXPLOG_pn54x_dev_ioctl, cmd=1074063617, arg=1
TML_LOGLEVEL=(0x3)

NxpExtns:  pn544_enable power on
 GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure

NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     pn54x_dev_ioctl, cmd=1074063617, arg=0
phTmlNfc_i2c_reset(), VEN level 0pn544_disable power off

[... 938 lines cut ...]
NxpNciX:     len =  28 > 2103190C000101010201030105018001810182018301850106017001
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   4 > 41030100
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=4
NxpFunc:    nfaConnectionCallback: event= 30
NxpFunc:    nfaConnectionCallback: NFA_RF_DISCOVERY_STARTED_EVT: status = 0
NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

Waiting for a Tag/Device...
```

Is there something wrong in it ?

jackofheartss, 2019-04-02:
If I look into the main.c of the app, this is what happens when I launch "nfcDemoApp poll" : 

- It goes into the _WaitDeviceArrival_ function
- Then into the _do while_ loop
- Then into the  _else if(eDevState_PRESENT != g_DevState)_
- Then inside _framework_WaitMutex(g_devLock, 0)_

I was looking for the file where  _framework_WaitMutex()_ is defined in order to continue my investigations but I cannot find it. So I don't know what is happening at this point.

jgeslin, 2019-04-02:
_framework_WaitMutex_ is defined in tools.c (same level than main.c).

From the logs, Initialization run smoothly as expected. The Application is then just waiting for event signaling the discovery of contactless device (like a card).
It may be because the NFC antenna is not supplied to generate external field and to discover the tag.
Are you using OM5578 demo kit or directly connecting PN7150 IC ?
If using OM5578 what is the voltage level on VANT pin (it must be above 3V (ideally 5V) ?

jackofheartss, 2019-04-02:
Thanks.
I have directly connected the PN7150 with an I2C bus (based on the circuitry page 6 fig. 2 : https://www.nxp.com/docs/en/application-note/AN11756.pdf ) to a DART-6UL.

jgeslin, 2019-04-02:
What the level of VBAT1 pin ?
Do you see signal on the antenna pins ?

jackofheartss, 2019-04-02:
The level on VBAT1 pin is 3.38 V.

I can see the oscillations of 13.56 MHz on the RX pins if I approach an NFC reader from the PN7150.

But if I poll, I don't see anything on RX or TX with a NFC tag near the antenna.

jgeslin, 2019-04-03:
Using a scope, as depicted in https://www.nxp.com/docs/en/application-note/AN11755.pdf chapter 5.2.1.2 Fig 37, do you see regular polling signal as shown in https://www.nxp.com/docs/en/application-note/AN11757.pdf Fig 2 or Fig 4 ?

jackofheartss, 2019-04-15:
it turns out I had no signal on my cristal (NX2016SA) : it was wrongly soldered (rotated by 90 degrees). This is because the PCB footprint is wrong on its datasheet : https://www.ndk.com/images/products/catalog/c_NX2016SA_e.pdf

Thank you for your support, it helped me finding the issue. The topic can be closed

closed 2019-04-15 by jackofheartss