NXPNFCLinux/linux_libnfc-nci #97: issue #26 not fix into R2.4

OSContributor, 2019-03-29:
This issue is linked to "infinite loops when no NFC device (100% CPU) #26".
It seems to me that this issue is not fixed into R2.4

Indeed, in https://github.com/NXPNFCLinux/linux_libnfc-nci/pull/27 you mention that the R2.4 includes the fix for #26 .

After code analysis, the patch does not seem to be applied correctly: 
https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/src/halimpl/pn54x/hal/phNxpNciHal.c#L951 shall be at the same level as https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/src/halimpl/pn54x/hal/phNxpNciHal.c#L965

In the current implementation, it is not possible the cover the code into https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/src/halimpl/pn54x/hal/phNxpNciHal.c#L951 due to https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/src/halimpl/pn54x/hal/phNxpNciHal.c#L922

May you check this issue and apply the fix in future release ?

Regards
OSContributor

jgeslin, 2019-04-01:
Indeed !
Now fix with commit https://github.com/NXPNFCLinux/linux_libnfc-nci/commit/42bd8edada46ea727434af07423813589141527b

Thanks,
Jeremy.

jgeslin, 2019-09-06:
Closing since fixed in release R2.4.1

closed 2019-09-06 by jgeslin