NXPNFCLinux/linux_libnfc-nci #98: nfcDemoApp poll errno: 5

JDavidson2019, 2019-04-21:
So I recently attempted to configure PN7120s on my Raspberry Pi 3 B+, running a brand new version of Raspbian stretch (4.14). I configured the NFC reader using the alt configuration for linux_libnfc-nci. The exact protocol I followed is from the NXP forums and is located [here.](https://community.nxp.com/docs/DOC-341231) However, when I run the poll option on the nfcDemoApp, I get the following debug output:

--------------------------------------------------------------------------------------------------------
pi@raspberrypi:~/linux_libnfc-nci $ ./nfcDemoApp poll > debug.txt
NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

NxpTml:     NFC - Assign IO pins

NxpTml:     Pin /sys/class/gpio/gpio23

NxpTml:     System already has pin /sys/class/gpio/gpio23

NxpTml:     Direction /sys/class/gpio/gpio23/direction

NxpTml:     Pin 23 now an input

NxpTml:     Edge /sys/class/gpio/gpio23/edge

NxpTml:     Edge-string rising - 6

NxpTml:     Pin 23 trigger on rising

NxpTml:     Value /sys/class/gpio/gpio23/value

NxpTml:     Pin /sys/class/gpio/gpio24

NxpTml:     System already has pin /sys/class/gpio/gpio24

NxpTml:     Direction /sys/class/gpio/gpio24/direction

NxpTml:     Pin 24 now an output

NxpTml:     Pin 24 now off

NxpTml:     NFCHW - open I2C bus - /dev/i2c-1

NxpTml:     NFC - open I2C device - 0x28

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Read requested.....

NxpTml:     NxpHal:     thread started
NxpTml:     PN54X - Write requested.....

thread startedg I2C Read.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 5
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
NxpHal:     PN54X Reset - SUCCESS

NxpHal:     phNxpNciHal_write failed for hal ext
NxpHal:     Force FW Download, NFCC not coming out from Standby
NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=6
------------------------------------------------------------------------------------------------

And the app just hangs once this point is reached.

jgeslin, 2019-04-23:
It may be because of the I2C port access rights.
What gives the following commands ?
 `$ ls -als /dev/i2c-1`

JDavidson2019, 2019-04-25:
I get the following readout:

<code> 0 crw-rw---- 1 root i2c 89,  1 Apr 24 20:05 /dev/i2c-1 </code>

jgeslin, 2019-04-25:
To check if this is related to I2C access rights, please run:
`$ sudo chmod 666 /dev/i2c-1`

Then run the demo app again.

jgeslin, 2019-04-26:
I've just run the same in my side with Raspbian stretch (4.14) and it works fine.
Can you please check PN7120 signals voltage level (accessible on the connectors located on the top layer of OM5577):
 - VBAT
 - I2C_SCL
 - I2C_SDA
 - IRQ
 - VEN

JDavidson2019, 2019-04-26:
Thank y I u! So I tried the command and it did not work. I even tried
getting a PI 3B and trying the demo image which has been reported to work
and got that's same behavior. I do apologize, but I am kinda new to
Raspberry PI, how might I check the signal level on those pins? (I know
where those pins are, I just don't I now how to check the output. Thanks!

jgeslin, 2019-04-26:
Just use a voltage meter to measure the voltage level of those signals.
- VBAT should be 3.3V
- I2C_SCL and I2C_SDA should be 3.3V (RPi internal pull-up)
- IRQ should be 0V
- VEN should be 3.3V