NXPNFCLinux/linux_libnfc-nci #100: NfcService Init Failed

mmar2228, 2019-05-13:
Dear all,

I have configured linux_libnfc-nci with --enable-alt flag. The phTmlNfc_alt.h file has been set up as follows:

#define I2C_BUS "/dev/i2c-0"
#define I2C_ADDRESS 0x28
#define PIN_INT 8
#define PIN_ENABLE 9

Note that pn5xx_i2c driver has not been used. 

When I start the demo: ./nfcDemoApp poll, I get the following error message:

CUSTOM> too many i2c read errors: ABORTING...
NfcService Init Failed
Leaving ...

Any idea about the reason? Should I configure the GPIO as read mode or do any additional setting?

Thanks you very much

jgeslin, 2019-05-13:
Hi,

Which CPU board are you using ?

Regards,
Jeremy.

mmar2228, 2019-05-13:
I copy the output of "cat /proc/cpuinfo" command:

$cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 16.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 16.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Freescale i.MX7 Dual (Device Tree)
Revision        : 0000
Serial          : 0000000000000000

Regards,

MMar

jgeslin, 2019-05-13:
Any chance that you could do an I2C trace (ideally using I2C spy) of the issue ?

jgeslin, 2019-05-13:
Do you have another NFC demoboard (OM5578 or OM5577) you could try ?

mmar2228, 2019-05-13:
I am newbie on this issue. I am using Linux over a SMARC device. I am looking for a command to spy the I2C trace, which is unknown for me.
If I use "i2cdetect -y 0" I get the following output:

 i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- UU -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: UU UU UU UU UU UU UU UU -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Does it mean that 0x28 is not the correct address?

jgeslin, 2019-05-13:
I2C spy would allow to see the I2C communication from the external side (between CPU and NFC device).
Can you at run "dmesg" after getting the issue an publish the output here ?

kanlll, 2019-05-14:
Hi MMar,

Actually I also met this issue before, but it finally was found as a connection issue. After I pulled and plugged it back, the demo worked as expected. Would you please check if that is the same case on your side?

Regards,
Kan

mmar2228, 2019-05-14:
I publish here the output of "dmesg" after trying "./nfcDemoApp poll". I will also try to pull and plug the module, and I will tell you. Thanks for your help!

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.11-1.0.1_smarc+g032ad45 (vem@vem) (gcc version 6.2.0 (GCC) ) #2 SMP PREEMPT Thu Mar 7 08:44:34 CET 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Embedian i.MX7D SMARC 2.0 Module
[    0.000000] Reserved memory: created CMA memory pool at 0xac000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat 80d67d80, node_mem_map ab759000
[    0.000000]   Normal zone: 2048 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 262144 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @ab71d000 s26188 r8192 d22964 u57344
[    0.000000] pcpu-alloc: s26188 r8192 d22964 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260096
[    0.000000] Kernel command line: console=ttymxc5,115200 root=/dev/mmcblk2p2 ro rootfstype=ext4 rootwait fixrtc
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 697784K/1048576K available (8192K kernel code, 418K rwdata, 2812K rodata, 1024K init, 460K bss, 23112K reserved, 327680K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xc0800000 - 0xff800000   (1008 MB)
                   lowmem  : 0x80000000 - 0xc0000000   (1024 MB)
                   pkmap   : 0x7fe00000 - 0x80000000   (   2 MB)
                   modules : 0x7f000000 - 0x7fe00000   (  14 MB)
                     .text : 0x80008000 - 0x80900000   (9184 kB)
                     .init : 0x80c00000 - 0x80d00000   (1024 kB)
                     .data : 0x80d00000 - 0x80d68b60   ( 419 kB)
                      .bss : 0x80d6a000 - 0x80ddd0f4   ( 461 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 8.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 440795202120 ns
[    0.000008] sched_clock: 56 bits at 8MHz, resolution 125ns, wraps every 2199023255500ns
[    0.000019] Switching to timer-based delay loop, resolution 125ns
[    0.000388] Ignoring duplicate/late registration of read_current_timer delay
[    0.000405] clocksource: mxc_timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
[    0.002321] Console: colour dummy device 80x30
[    0.002340] Calibrating delay loop (skipped), value calculated using timer frequency.. 16.00 BogoMIPS (lpj=80000)
[    0.002355] pid_max: default: 32768 minimum: 301
[    0.002449] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002459] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003174] CPU: Testing write buffer coherency: ok
[    0.003467] CPU0: update cpu_capacity 1024
[    0.003479] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003510] Setting up static identity map for 0x80100000 - 0x80100058
[    0.170382] CPU1: update cpu_capacity 1024
[    0.170390] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.170486] Brought up 2 CPUs
[    0.170501] SMP: Total of 2 processors activated (32.00 BogoMIPS).
[    0.170507] CPU: All CPU(s) started in SVC mode.
[    0.171201] devtmpfs: initialized
[    0.184622] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.185021] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.196059] pinctrl core: initialized pinctrl subsystem
[    0.197243] NET: Registered protocol family 16
[    0.204892] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.230331] cpuidle: using governor ladder
[    0.260358] cpuidle: using governor menu
[    0.271064] DDR type is DDR3!
[... 358 lines cut ...]
[    3.640592] systemd[1]: Set hostname to <imx7dsmarc>.
[    3.839162] systemd[1]: sysinit.target: Found ordering cycle on sysinit.target/start
[    3.845733] systemd[1]: sysinit.target: Found dependency on alignment.service/start
[    3.852165] systemd[1]: sysinit.target: Found dependency on sysinit.target/start
[    3.858271] systemd[1]: sysinit.target: Breaking ordering cycle by deleting job alignment.service/start
[... cut ...]
[    7.130244] Atheros 8035 ethernet 30be0000.etherne:06: attached PHY driver [Atheros 8035 ethernet] (mii_bus:phy_addr=30be0000.etherne:06, irq=-1)
[    7.190373] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.310925] Atheros 8035 ethernet 30be0000.etherne:07: attached PHY driver [Atheros 8035 ethernet] (mii_bus:phy_addr=30be0000.etherne:07, irq=-1)
[    7.390889] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[    9.280656] fec 30be0000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[    9.287308] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

mmar2228, 2019-05-15:
Pulling and plugging the module doesn`t fix the issue. I am still trying to figure out what is happening...

jgeslin, 2019-05-15:
Please enable NFC library traces : in /usr/local/etc/libnfc-nxp-init.conf file set all NXPLOG_xxx_LOGLEVEL flags to 0x03 (instead of 0x00).
Run the app "nfcDemoApp poll 2>log.txt" and share the obtained log.txt file here.

mmar2228, 2019-05-15:
I print here the content of log.txt:

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

NxpTml:     NFC - Assign IO pins

NxpTml:     Pin /sys/class/gpio/gpio8

NxpTml:     Create pin /sys/class/gpio/gpio8

NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     Direction /sys/class/gpio/gpio8/direction

NxpTml:     Pin 8 now an input

NxpTml:     Edge /sys/class/gpio/gpio8/edge

NxpTml:     Edge-string rising - 6

NxpTml:     Pin 8 trigger on rising

NxpTml:     Value /sys/class/gpio/gpio8/value

NxpTml:     Pin /sys/class/gpio/gpio9

NxpTml:     Create pin /sys/class/gpio/gpio9

NxpTml:     Direction /sys/class/gpio/gpio9/direction

NxpTml:     Pin 9 now an output

NxpTml:     Pin 9 now off

NxpTml:     NFCHW - open I2C bus - /dev/i2c-0

NxpTml:     NFC - open I2C device - 0x28

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     thread started
NxpTml:     PN54X - Read requested.....

[... 799 lines cut ...]
NxpExtns:   find found NXP_NFC_MAX_EE_SUPPORTED=(0x0)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXP_NFC_MAX_EE_SUPPORTED=(0x0)

NxpFunc:    Configure: Overriding NFA_EE_MAX_EE_SUPPORTED to use 0
NxpFunc:    nativeNfcManager_doInitialize: fail nfa enable; error=0x0
NxpFunc:    NfcAdaptation::Finalize: enter
NxpFunc:    Nfc: GKI TASK_DEAD received. exit thread 3...

NxpFunc:    NfcAdaptation::Finalize: exit
NxpFunc:    nativeNfcManager_doInitialize: nfc enabled = 0
NxpFunc:    nativeNfcManager_doInitialize: exit
NxpFunc:    nativeNfcSnep_stopServer:
NxpFunc:    nativeNfcSnep_stopServer: Nfc not initialized.
NxpFunc:    nativeNfcManager_disableDiscovery: enter;
NxpFunc:    nativeNfcManager_disableDiscovery: Nfc not initialized.
NxpFunc:    nativeNfcManager_disableDiscovery: exit
NxpFunc:    nativeNfcSnep_deregisterClientCallback:
NxpFunc:    nativeNfcSnep_deregisterClientCallback: Nfc not initialized.
NxpFunc:    RoutingManager::RoutingManager(): default route is 0x00

NxpFunc:    nativeNfcManager_doDeinitialize: enter
NxpFunc:    nativeNfcManager_doDeinitialize: Nfc not initialized.
NxpFunc:    NfcAdaptation::NFCA_TASK: exit
NxpFunc:    ~RoutingManager:~RoutingManager()

kanlll, 2019-05-16:
Hi MMar,

I also pulled OM5578/PN7150S off the OM29110RPI board and plugged it back , but I am not sure if you did it in the same way. Would you please clarify?

Thanks for your patience!

Regards,
Kan

Caution: EXT Email

kanlll, 2019-05-16:
Hi MMar，

I also pulled OM5578 off the OM29110RPI board and plugged it back, did you do it in the same way? Please kindly clarify. 

Regards,
Kan

mmar2228, 2019-05-16:
Oh, you are right. I am not using the OM29110RPI board. I connect OM5578/PN7150 to a customized board., because it will be integrated in an embedded system

jgeslin, 2019-05-16:
Please run the following command and give results:
`$ cat /sys/class/gpio/gpio8/direction; cat /sys/class/gpio/gpio8/value; i2cdetect -y 0; i2cdectect -y 0`

mmar2228, 2019-05-16:
The gpio8 direction is "in", the gpio8 value is "1" and i2cdetect -y 0 gives the followng output:

i2cdetect -y 0
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- UU -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: UU UU UU UU UU UU UU UU -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

As I asked in a previous message: Does it mean that 0x28 is not the correct address?

On the other hand, I have realize that my system uses special names for GPIOs. I mean, if I want to act over gpio9, I should to change the direction and values in /sys/class/gpio/gpio97, according to the user manual. In this way, gpio8 uses the /sys/class/gpio/gpio38 folder. What changes should I do in order to consider it?

Very gratefully,

MMar

mmar2228, 2019-05-16:
Now previous error has disappear ("CUSTOM> too many i2c read errors: ABORTING...). However I am not still able to get the "Waiting for a Tag/Device" message. I copy here the log.txt after doing "./nfcDemoApp poll 2>log.txt". Thannks!! 

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

NxpTml:     NFC - Assign IO pins

NxpTml:     Pin /sys/class/gpio/gpio38

NxpTml:     System already has pin /sys/class/gpio/gpio38

NxpTml:     Direction /sys/class/gpio/gpio38/direction

NxpTml:     Pin 38 now an input

NxpTml:     Edge /sys/class/gpio/gpio38/edge

NxpTml:     Edge-string rising - 6

NxpTml:     Pin 38 trigger on rising

NxpTml:     Value /sys/class/gpio/gpio38/value

NxpTml:     Pin /sys/class/gpio/gpio97

NxpTml:     System already has pin /sys/class/gpio/gpio97

NxpTml:     Direction /sys/class/gpio/gpio97/direction

NxpTml:     Pin 97 now an output

NxpTml:     Pin 97 now off

NxpTml:     NFCHW - open I2C bus - /dev/i2c-0

NxpTml:     NFC - open I2C device - 0x28

NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     thread started
NxpTml:     PN54X - Read requested.....

[... 130 lines cut ...]

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
NxpHal:     PN54X Reset - SUCCESS

NxpHal:     phNxpNciHal_write failed for hal ext
NxpHal:     Force FW Download, NFCC not coming out from Standby
NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=6

mmar2228, 2019-05-16:
And here the GPIOs info:

gpio38 -> cat value gives "0"
            -> cat direction gives "ini"
gpio97 -> cat value gives "1"
            -> cat direction gives "out"

jgeslin, 2019-05-16:
Please check level of the signal level (using scope or voltage meter) when the app is running :
- I2C SDA
- I2C SCL
- VEN
- VBAT/PVDD

mmar2228, 2019-05-17:
Sure,

I2C SCL is not oscillating (it is high all time), I don't know why. However, when GPIOs pins were bad adjusted (9 and 8 instead of the real 97 and 38), I2C SCL wave looks fine.

On the other hand, VEN and VBAT/PVDD is 3,3 V.

jgeslin, 2019-05-17:
Seems to be a pin configuration issue.
Can you share your DTS file  as well as the connection list between iMX7 pins and OM5578 ?

mmar2228, 2019-05-17:
The connections are the following:

NXP NFC GPIO 0 <---> SMARC GPIO8 (IRQ)
NXP NFC GPIO 1 <---> SMARC GPIO9 (VEN)
NXP NFC i2C_SDA <---> SMARC I2C1_SDA
NXP NFC i2C_SCL <---> SMARC I2C1_SCL

And here the DTS file:

/*
 * Copyright (C) 2015 Embedian, Inc.
 *
 * This file is dual-licensed: you can use it either under the terms
 * of the GPL or the X11 license, at your option. Note that this dual
 * licensing only applies to this file, and not this project as a
 * whole.
 *
 *  a) This file is free software; you can redistribute it and/or
 *     modify it under the terms of the GNU General Public License as
 *     published by the Free Software Foundation; either version 2 of the
 *     License, or (at your option) any later version.
 *
 *     This file is distributed in the hope that it will be useful,
 *     but WITHOUT ANY WARRANTY; without even the implied warranty of
 *     MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *     GNU General Public License for more details.
 *
 * Or, alternatively,
 *
 *  b) Permission is hereby granted, free of charge, to any person
 *     obtaining a copy of this software and associated documentation
 *     files (the "Software"), to deal in the Software without
 *     restriction, including without limitation the rights to use,
 *     copy, modify, merge, publish, distribute, sublicense, and/or
 *     sell copies of the Software, and to permit persons to whom the
 *     Software is furnished to do so, subject to the following
 *     conditions:
 *
 *     The above copyright notice and this permission notice shall be
 *     included in all copies or substantial portions of the Software.
 *
 *     THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 *     EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 *     OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 *     NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 *     HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 *     WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 *     FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 *     OTHER DEALINGS IN THE SOFTWARE.
 */

/dts-v1/;

#include "imx7d.dtsi"

/ {
	model = "Embedian i.MX7D SMARC 2.0 Module";
	compatible = "fsl,imx7d-smarcfimx7", "fsl,imx7d";

	memory {
		reg = <0x80000000 0x80000000>;
	};

        aliases {
               ethernet1 = &fec2;      		/* Let eth1addr mac address pass from U-Boot EEPROM */
        };

	regulators {
		compatible = "simple-bus";
		#address-cells = <1>;
[... 1008 lines cut ...]
				MX7D_PAD_SD3_DATA4__SD3_DATA4		0x5a
				MX7D_PAD_SD3_DATA5__SD3_DATA5		0x5a
				MX7D_PAD_SD3_DATA6__SD3_DATA6		0x5a
				MX7D_PAD_SD3_DATA7__SD3_DATA7		0x5a
				MX7D_PAD_SD3_STROBE__SD3_STROBE		0x1a
			>;
		};

		pinctrl_usdhc3_200mhz: usdhc3grp_200mhz {
			fsl,pins = <
				MX7D_PAD_SD3_CMD__SD3_CMD		0x5b
				MX7D_PAD_SD3_CLK__SD3_CLK		0x1b
				MX7D_PAD_SD3_DATA0__SD3_DATA0		0x5b
				MX7D_PAD_SD3_DATA1__SD3_DATA1		0x5b
				MX7D_PAD_SD3_DATA2__SD3_DATA2		0x5b
				MX7D_PAD_SD3_DATA3__SD3_DATA3		0x5b
				MX7D_PAD_SD3_DATA4__SD3_DATA4		0x5b
				MX7D_PAD_SD3_DATA5__SD3_DATA5		0x5b
				MX7D_PAD_SD3_DATA6__SD3_DATA6		0x5b
				MX7D_PAD_SD3_DATA7__SD3_DATA7		0x5b
				MX7D_PAD_SD3_STROBE__SD3_STROBE		0x1b
			>;
		};
	};
};

jgeslin, 2019-05-17:
Do you mean ?

NXP NFC GPIO 0 <---> SMARC GPIO8 (IRQ)      --> SMARC PIN P116 = iMX7 M21
NXP NFC GPIO 1 <---> SMARC GPIO9 (VEN)     --> SMARC PIN P117 = iMX7 L4
NXP NFC i2C_SDA <---> SMARC I2C1_SDA        --> SMARC PIN P122 = iMX7 K1 
NXP NFC i2C_SCL <---> SMARC I2C1_SCL         --> SMARC PIN P121 = iMX7 J2

mmar2228, 2019-05-17:
Yes, it is exactly

jgeslin, 2019-05-17:
Are you sure if /dev/i2c-0 is mapped to I2C1 ?
Just to test, connect OM5578 VEN to 3.3V (instead of GPIO) to fix the level on the pin, and run twice (in a raw) i2cdetect command:
`$ i2cdetect -y 0; i2cdetect -y0`

mmar2228, 2019-05-17:
We have also a level changer to et 3.3V instead of 1.8 V between the OM5578 and SMARC

mmar2228, 2019-05-17:
According to the manual, SMARC PIN 122 and 121 are I2C1_SDA and I2C1_SCL.
In addtition, manual indicates there are I2C buses from I2C1 to I2C4. And "ls /dev/i2c*" gives me from /dev/i2c-0 to /dev/i2c.-3. So, I2C1 must be the /dev/i2c-0 device.

However, i2cdetect -y 0; i2cdetect -y 0 gives me the same as previously I indicated.

i2cdetect -y 1; i2cdetect -y 1 gives me:

1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

It is logical since I2C2 has a device with 0x50 address.

jgeslin, 2019-05-17:
when you run `i2cdetect -y 0` do you see I2C1_SCL toggling between 0 and 3.3V ?

mmar2228, 2019-05-17:
Yes, I see it

jgeslin, 2019-05-17:
OM5578 fixes (by HW) the I2C slave address to 0x28.
if PN7150 is supplied (VBAT = 3.3V) and enabled (VEN = 3.3V), it shall acknowledge I2C requests (assuming I2C signals level are 3.3V) with address 0x28.
If it doesn't, the board may be broken. Do you have the possibility to check with another one or to verify it with another environment (Raspberry Pi for instance) ?

mmar2228, 2019-05-17:
I will try it. Thanks you very much

mmar2228, 2019-05-17:
I have tried with another OM5578 and the result is the same. When I run the app, at first I can see I2C_SCL and I2C_SDA toggling, but then they both hold high level. VEN and VBAT are high but IRQ is set to zero

mmar2228, 2019-05-19:
I can confirm you that board is not broken. I have tried the same  with Raspberry Pi and 'Waiting for a Tag/Service...' message is shown. I have also checked that i2cdetect -y 1 (in this case we use i2c-1 bus) only shows the OM5578 device (it lists 0x28 address) once demo app has been run. After booting process i2cdetect does not find any device. So, I think my SMARC does not list the OM5578 device with i2cdetect command because demo app has never worked correctly. But the unresolved question is: why?
I appreciate your time and priceless help.

mmar2228, 2019-05-20:
Comparing both cases (SMARC failed and successfully Raspberry Pi) I notice that when app is starting, IRQ pin holds the zero value in SMARC but it toggles in Raspberry Pi. Maybe it can be the reason... What value should hold that signal? Toggling between zero and 3,3V?

jgeslin, 2019-05-20:
PN7150 raises IRQ line up to indicates it has something to send to the host, so the behavior observed on SMARC is expected since PN7150 seems not to handle any I2C request (not acknowledging command on address 0x28).

The next step would be to make a scope diagram of the I2C communication (+ ven and vbat on the same scope) when running demo app.

Caution: EXT Email

mmar2228, 2019-05-21:
[image]
[image]

I attach the i2c communication when app is starting, both in Raspberry Pi and SMARC cases. I have also checked than in both scenarios VEN and VBAT pins hold to 3.3V.

I publish here the log.txt in SMARC ('Waiting for a Tag/Service...' is not shown). 

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure Alternative NFC

NxpTml:     NFC - Assign IO pins

NxpTml:     Pin /sys/class/gpio/gpio38

NxpTml:     Create pin /sys/class/gpio/gpio38

NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     Direction /sys/class/gpio/gpio38/direction

NxpTml:     Pin 38 now an input

NxpTml:     Edge /sys/class/gpio/gpio38/edge

NxpTml:     Edge-string rising - 6

NxpTml:     Pin 38 trigger on rising

NxpTml:     Value /sys/class/gpio/gpio38/value

NxpTml:     Pin /sys/class/gpio/gpio97

NxpTml:     Create pin /sys/class/gpio/gpio97

NxpTml:     Direction /sys/class/gpio/gpio97/direction

NxpTml:     Pin 97 now an output

NxpTml:     Pin 97 now off

NxpTml:     NFCHW - open I2C bus - /dev/i2c-0

NxpTml:     NFC - open I2C device - 0x28

NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     thread started
NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
PN54X - Tml Writer Thread Running................

NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     _i2c_write() errno : 6
NxpTml:     PN54X - Error in I2C Write.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write error status = 0x1ff
NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode (max count = 0x6)
PN54X - Tml Writer Thread Running................

NxpHal:     PN54X Reset - SUCCESS

NxpHal:     phNxpNciHal_write failed for hal ext
NxpHal:     Force FW Download, NFCC not coming out from Standby
NxpFunc:    NfcAdaptation::HalDeviceContextCallback: event=6

mmar2228, 2019-05-21:
I get similar waveform when I remove the PN7150 module from the board. Sadly, I am afraid PN7150 is not identify by SMARC device. But I have no idea the reason. Any test could I do?

jgeslin, 2019-05-21:
What does give "i2cdetect -y 0" when OM5578 is not connected ?

mmar2228, 2019-05-21:
"i2cdetect -y 0" gives me the same result both when OM5578 is connected and when it is not connected to SMARC. It does not find the 0x28 address

In the Raspberry Pi case, when the OM5578 is connected and the app has been run, "i2cdetect -y 1" gives me the 0x28 address. And if I disconnect the OM5578 from Raspi I get none address, as expected.

mmar2228, 2019-05-21:
It seems all OM5578 input are correct (I2C_SCL, I2C_SDA, VEN, VBAT). However, it is not alive on SMARC

mmar2228, 2019-05-21:
[image]
[image]

However I2C communication in Raspi and SMARC looks different. Could be it useful information to get the problem solved? How should be the I2C communication scope?

Thanks very very much!

jgeslin, 2019-05-22:
Seems we are missing the real beginning of the I2C communication on SMARC device. 
Can you do again the scope diagram triggering on the first SCL falling edge (single trigger on scope) ?

mmar2228, 2019-05-22:
Yes, I noticed that it after sending you the scopes. Here the correct one for SMARC.
[image]

jgeslin, 2019-05-22:
Can you add VEN and VBAT on this scope diagram ? Eventually in place of SDA in case you only have 2 inputs.

mmar2228, 2019-05-22:
[image]
[image]

First image are VBAT and SCL signals. Second one are VEN and SCL signals. Both VBAT and VEN are holding high value (3.3V)

jgeslin, 2019-05-22:
That's really weird !
You have exactly same behavior of the host on SMARC and RPI (host is sending I2C Write request at address 0x28) but in the case of RPI, PN7150 does acknowledge by lowering SDA during the 9th bit while it does not on SMARC. The only reason why it wouldn't acknowledge is because it did not boot (no power or HW reset) but the level of VBAT and VEN looks ok.

Can you please make a new I2C trace on SMARC to see the first exchanges (more than the first byte) ?

mmar2228, 2019-05-22:
[image]
[image]
[image]
I attach 3 scopes. When I do the scope bigger, resolution decreases and that's why it seems the 0x28 is not sent, but high resolutions gives the real 0x28 address.

jgeslin, 2019-05-22:
So can you capture whole sequence (as on the picture #2) and make 3 picture zooming on each byte sequence (first is already picture #1)  ?

Caution: EXT Email

[scope_0]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F35975894%2F58177952-97382d00-7ca5-11e9-8156-42523c3a4ca0.png&data=02%7C01%7Cjeremy.geslin%40nxp.com%7C322887278aaa4852ca5e08d6deb91637%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C636941283850278585&sdata=VvdrKgC6ARrrz7si13iLZAIi1sNMaVSbmbcHSUM9jOE%3D&reserved=0>
[scope_2]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F35975894%2F58177958-999a8700-7ca5-11e9-9d0b-038a4361b999.png&data=02%7C01%7Cjeremy.geslin%40nxp.com%7C322887278aaa4852ca5e08d6deb91637%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C636941283850278585&sdata=1D6QZq5UHW16A40V0tVkut3fU3quXpT%2By1yMLAzi1n0%3D&reserved=0>
[scope_3]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F35975894%2F58177963-9b644a80-7ca5-11e9-8b16-f781cd312ac5.png&data=02%7C01%7Cjeremy.geslin%40nxp.com%7C322887278aaa4852ca5e08d6deb91637%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C636941283850288582&sdata=3RTWN%2FLVYEGDdgvyYJ4V1nfLog%2FWQuVENfU9q6FTKP8%3D&reserved=0>

mmar2228, 2019-05-22:
I am going to be crazy!
Sometimes I get the following scope:
[image]

And other times I get the other one:
[image]

I have split the 3 bytes:

[image]
[image]
[image]

mmar2228, 2019-05-22:
First scope (with 0x28 addresss) and no reply is what I also get with my second OM5578 module that fails both in Raspi and SMARC

jgeslin, 2019-05-22:
> 
> 
> First scope (with 0x28 addresss) and no reply is what I also get with my second OM5578 module that fails both in Raspi and SMARC

Ok, that's at least explain the weird scope diagram with I2C request to other address.

Then, to isolate the problem, any chance you could disconnect the 2 other devices on the bus or disable the related module ?

mmar2228, 2019-05-23:
These devices are included in the SMARC board. I can not disconnect them and if I could it I don't know the consequences of doing it. However, usually I get the waveform with 0x28 address request and no response. 

I also checked that the second request to 0x28 address gives a noisy I2C communication.

[image]

First byte request:
[image]

After this first byte, there is no reply by PN1750, I2C and SDA hold high values.

Second byte request:
[image]

In addition, after noise, 0x28 address request come back, but noise appers again:

[image]

And finally I2C SCL and I2C SDA hold high values again:
[image]

mmar2228, 2019-06-18:
Dear Jeremy,

Firstly, I would like to be grateful for your attentive help with my problem. I have solved it by changing the level shift device. It seems it was not compatible. I apologize for the inconvenience caused.

Thanks very much for your help. Best regards,

M del Mar

closed 2019-06-18 by jgeslin