NXPNFCLinux/linux_libnfc-nci #102: Changing sectors on a Type 2 tag

kevinfields-canopygrowth, 2019-06-13:
I have a Type 2 NFC tag that supports multiple sectors, and I am trying to find a way to use the library to perform a sector change so that I can read and write data on the sectors that contain user space.

I've attempted to use `nfcTag_transceive` to perform the 2-step sector select command, like the datasheet for my NFC tag type indicates; however, this results in the second step command timing out. After some investigation, I couldn't find a public API that I could use from the available headers that would allow me to perform a sector change. What I did find was the `NFA_RwT2tSectorSelect` method within https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/src/libnfc-nci/nfa/include/nfa_rw_api.h, but this doesn't appear to be getting called anywhere within the library.

Is there a way with the library to be able to perform a sector change on a Type 2 tag, or will modifications to the library be required?

jgeslin, 2019-06-20:
Hi,

Indeed, the current version doesn't allow SECTOR_SELECT command because the second command packet is implicitly Acknowledge by a non response of the tag. This generates timeout of the transceive operation and this is interpreted as a Tag lost.

Please find attached a patch which prevent timeout when transceive function is called with timeout parameter value set to 0: [patch_SectorSelect.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3309836/patch_SectorSelect.txt).

Then you can select a sector in your app using transceive API as for instance:
```
 unsigned char SecSel1Cmd[] = {0xC2, 0xFF};
 unsigned char SecSel2Cmd[] = {/* sector */ 0x01, 0x00, 0x00, 0x00};
 unsigned char Resp[16];

 res = nfcTag_transceive(handle, SecSel1Cmd, sizeof(SecSel1Cmd), Resp, sizeof(Resp), 500);
 res = nfcTag_transceive(handle, SecSel2Cmd, sizeof(SecSel2Cmd), Resp, sizeof(Resp), 0);

```
The first transceive call shall return ACK value (1 byte, '0x0A'), while the second call shall return empty (0 bytes) because of the passive ACK.

Regards,
Jeremy.

kevinfields-canopygrowth, 2019-06-21:
Hi Jeremy,

Thank you very much for the patch. It worked nicely.

Kind Regards,
Kevin

jgeslin, 2019-09-06:
Closing since fixed in R2.4.1 tag

closed 2019-09-06 by jgeslin