NXPNFCLinux/linux_libnfc-nci #103: NfcService Init Failed :LS1046AFRWY

saurabhchahan151, 2019-06-18:
Hi All,

I am working on LS1046AFRWY for NFC547.
so according to 1046AFRWY schmatics interrupt from microclick bus interrupt is going directly to IRQ07 to the processor.

Before this I have worked on LS1028ARDB for the same NFC547 module, I used this link for the driver( https://github.com/NXPNFCLinux/nxp-pn5xx.git ). In LS1028ARDB schematics interrupt from micro-click bus was going to GPIO interrupt controller. for this board NFC is working fine. Working case was IRQ_TYPE_EDGE_FALLING

I used same above link for driver code and modified according to schematic and board Reference Manual details, because no GPIO is using in LS1046AFRWY for interrupt. and I made DTS changes according to 1046AFRWY RM, in that for micro-click module internal irq is 178(IRQ_TYPE_EDGE_RISING). When i run "nfcDemoapp poll" interrupt is triggering to processor and able to i2c read/write from nfc547 module  to nfcapp by seeing dmesg log  but its giving :: 

NfcService Init Failed
Leaving ...

I compared the dmesg log from LS1028ARDB(working case) and LS1046AFRWY, it seems number of read/write and number of time read/write is mismatch.
And I also checked the buffer msg for both board, there is difference in msg. 
For the same nfc547 module for both the board showing different versions due to buffer mismatch
// ls1046afrwy
NxpHal:     NxpNci> FW Version: 0.2.2
NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x08
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN67T
NxpNciR:     len =  26 > 400117400117000000FFFFC045FB670880FFFF18C61008000202
    
// ls1028ardb
NxpHal:     NxpNci> FW Version: 8.2.2
NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x05
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN547C3
NxpNciR:     len =  26 > 40011700030E03000600010203818202C800FF02000405080202

Is there any dependency of GPIO in https://github.com/NXPNFCLinux/linux_libnfc-nci.git for "nfcDemoapp poll"?
OR what reason can be for this buffer message mismatch and for mismatch in number of i2c read/write between nfc module and nfcapp? 
  
I attached the log file below for both the board for dmesg and nfcapp DEBUG LOG:: 
[log_1046afrwy.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3300281/log_1046afrwy.txt)

jgeslin, 2019-06-18:
Hi,

"400117400117000000FFFFC045FB670880FFFF18C61008000202" is definitely not valid NFC Controller response. "40011700030E03000600010203818202C800FF02000405080202" you have on ls1028ardb is correct one.
It looks to me that you are having an issue with your I2C driver, this doesn't looks like related to the IRQ GPIO management. This leads to a fail in the NfcService initialization.
You can look to I2C trace (using I2C spy or scope) to confirm that what the NFC controller answers the same and valid answer but that it is incorrectly read by the I2C MPU driver. 

Regards,
Jeremy.

saurabhchahan151, 2019-06-19:
I am newbie.
I probed SDA line on microclick module and after running "nfcDemoapp  poll" i am getting burst of data (attached jpg below)
[image]
can you please suggest me how to debug exact data format for any read/write action?

jgeslin, 2019-06-19:
The idea behind scoping the I2C lines (you'll need SCL as well to sync) is to manually decode the frame and verify that the physical I2C answer to "200100" command.

I created a patch specific to you to overcome the corrupted answer and to see if next communcation exchanges also got corrupted. Please try it and share new log. 
[patch_LS1028ARDB.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3304931/patch_LS1028ARDB.txt)

saurabhchahan151, 2019-06-27:
Hi,
We are able  to work on nfcDemoApp by  "configure --enable-alt" . But one thing want to change,
after running the "nfcDemoApp poll", it will wait tag for 10 second. i want to change this timing. from where it can be controlled.

closed 2019-07-03 by saurabhchahan151

jgeslin, 2019-07-03:
How did you fix your issue ?

saurabhchahan151, 2019-07-15:
using alternate to nfc kernel driver by "configure --enable-alt" and did some changes:
1. src/halimpl/pn54x/tml/i2c/phTmlNfc_alt.h  => #define I2C_BUS  "/dev/i2c-4"  //as per board specific(LS1046AFRWY) 
2. src/halimpl/pn54x/tml/i2c/phTmlNfc_i2c.c => in function phTmlNfc_i2c_read() by adding extra sleep after read the data. it worked fine