NXPNFCLinux/linux_libnfc-nci #104: pn7150: nfcDemoApp fails with abort after nfaConnectionCallback: event= 30

moberhuber, 2019-06-28:
Hello,

I followed this guide : https://www.nxp.com/docs/en/application-note/AN11697.pdf to install the libnfc-nci onto an embedded Linux system. I'm using the "alt" mode for gpio / i2c and nfcDemoApp is running as root. I have hooked up the OM5578 demo kit to my custom board. When I do
```
./nfcDemoApp poll
```
it only prints 
```
CUSTOM> too many i2c read errors: ABORTING...
NfcService Init Failed
```
Enabling 0x03 logging as described in the application note, I do see communication like detecting firmware version etc, but then it seems to get stuck:
```
NxpHal:     NxpNci> FW Version: 10.1.10
NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=28
[...]
NxpExtns:   find found NXP_NFC_MAX_EE_SUPPORTED=(0x0)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXP_NFC_MAX_EE_SUPPORTED=(0x0)

NxpFunc:    Configure: Overriding NFA_EE_MAX_EE_SUPPORTED to use 0
NxpFunc:    RoutingManager::RoutingManager(): default route is 0x00

NxpFunc:    initialize:HOST_LISTEN_TECH_MASK=7;
NxpFunc:    RoutingManager::initialize(): try ee register
NxpFunc:    RoutingManager::nfaEeCallback: NFA_EE_REGISTER_EVT; status=0
NxpFunc:    nfaEeCallback: NFA_EE_DISCOVER_REQ_EVT; status=0x0; num ee=0
NxpFunc:    nativeNfcTag_registerNdefTypeHandler
NxpFunc:    RoutingManager::stackCallback: event=0x15
NxpFunc:    RoutingManager::stackCallback: NFA_CE_REGISTERED_EVT; status=0x0; h=0x301
NxpFunc:    ndefHandlerCallback: event=0, eventData=0x75672e0c
NxpFunc:    ndefHandlerCallback: NFA_NDEF_REGISTER_EVT; status=0x0; h=0x200
NxpFunc:    NfcTag::resetTechnologies
NxpFunc:    nativeNfcManager_doInitialize: NAME_POLLING_TECH_MASK mask=0xEF
NxpFunc:    nativeNfcManager_doInitialize: tag polling tech mask=0xEF
NxpFunc:    nativeNfcManager_doInitialize: P2P listen tech mask=0xC5
NxpFunc:    nativeNfcManager_doInitialize: nfc enabled = 1
NxpFunc:    nativeNfcManager_doInitialize: exit
NxpFunc:    nativeNfcSnep_registerClientCallback:
NxpFunc:    nfaSnepClientCallback: snepEvent= 0
NxpFunc:    nfaSnepClientCallback: NFA_SNEP_REG_EVT; Status: 0x0000

NxpFunc:    nfaSnepClientCallback: NFA_SNEP_REG_EVT; Client Register Handle: 0x0700

NxpFunc:    nativeNfcManager_enableDiscovery: enter; tech_mask = ef
NxpFunc:    nativeNfcTag_acquireRfInterfaceMutexLock: try to acquire lock
NxpFunc:    nativeNfcTag_acquireRfInterfaceMutexLock: sRfInterfaceMutex lock
NxpFunc:    nativeNfcManager_enableDiscovery: Disable p2pListening
NxpFunc:    enableP2pListening: enter isEnable: 0  mIsP2pListening: 0
NxpFunc:    enableP2pListening: exit; mIsP2pListening: 0
NxpFunc:    stopPolling_rfDiscoveryDisabled: disable polling
NxpFunc:    nfaConnectionCallback: event= 1
NxpFunc:    nfaConnectionCallback: NFA_POLL_DISABLED_EVT: status = 3
NxpFunc:    startPolling_rfDiscoveryDisabled: enable polling
NxpFunc:    startPolling_rfDiscoveryDisabled: wait for enable event
NxpFunc:    nfaConnectionCallback: event= 0
NxpFunc:    nfaConnectionCallback: NFA_POLL_ENABLED_EVT: status = 0
NxpFunc:    nativeNfcManager_enableDiscovery: Enable p2pListening
NxpFunc:    enableP2pListening: enter isEnable: 1  mIsP2pListening: 0
NxpFunc:    nfaConnectionCallback: event= 33
NxpFunc:    nfaConnectionCallback: NFA_SET_P2P_LISTEN_TECH_EVT
NxpFunc:    enableP2pListening: exit; mIsP2pListening: 1
NxpFunc:    RoutingManager::disableRoutingToHost() enter
NxpFunc:    RoutingManager::nfaEeCallback: NFA_EE_SET_PROTO_CFG_EVT; status=0x0
NxpFunc:    RoutingManager::nfaEeCallback: unknown event=9 ????
NxpFunc:    startRfDiscovery: is start=1
NxpFunc:    NfcAdaptation::HalWrite
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =  26 > 20021701291446666D010111020207FF03020003040164070103
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write successful status = 0x0
PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   5 > 4002020000
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=5
NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpFunc:    NfcAdaptation::HalWrite
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =  19 > 20021005320120300104310100380101500100
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   5 > 4002020000
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=5
NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpFunc:    NfcAdaptation::HalWrite
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   8 > 200205010002F401
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     NxpHal:     write successful status = 0x0
PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   5 > 4002020000
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_OK
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=5
NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpFunc:    NfcAdaptation::HalPrediscover
NxpFunc:    NfcAdaptation::HalWrite
NxpHal:     > Polling Loop Started
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =  22 > 21031309000101010201030105018001810106017001
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   4 > 41030100
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=4
NxpTml:     NxpFunc:    nfaConnectionCallback: event= 30
NxpFunc:    nfaConnectionCallback: NFA_RF_DISCOVERY_STARTED_EVT: status = 0
PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpFunc:    nativeNfcTag_releaseRfInterfaceMutexLock: sRfInterfaceMutex unlock
NxpFunc:    nativeNfcManager_enableDiscovery: exit
NxpFunc:    nativeNfcSnep_startServer:
NxpFunc:    startRfDiscovery: is start=0
NxpFunc:    NfcAdaptation::HalWrite
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   9 > 600006A00079DF0000
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpHal:     CORE_RESET_NTF received!
NxpHal:     phNxpNciHal_emergency_recovery: abort()
```

What could be going wrong here?

jgeslin, 2019-06-28:
Hi,

Can you do a touch on file "/usr/local/etc/libnfc-nxp-pn548.conf" (this will trigger a full initialization of the libnfc-nci library) and provides logs of the first startup afterwards ?

Regards,
Jeremy.

moberhuber, 2019-06-28:
Thanks Jeremy, respective
[nfcDemo2.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3339382/nfcDemo2.log) is attached and on https://we.tl/t-KXSY5S7Mwt .

jgeslin, 2019-06-28:
How did you connected VANT ?
Can you do a test by updating /usr/local/etc/libnfc-nxp-pn548.conf with:
NXP_EXT_TVDD_CFG=0x01 instead of NXP_EXT_TVDD_CFG=0x02 ?

moberhuber, 2019-06-28:
Thanks Jeremy, I've updated the TVDD_CFG, result 
[nfcDemo3.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3339639/nfcDemo3.log) doesn't look much different to me. I don't have the device in front of me right now (accessing it remotely) but I could answer hardware/electronics questions on Monday...

moberhuber, 2019-07-01:
Hi Jeremy,

> How did you connected VANT ?

Turns out we use external source at 3.3V and we measure 3.3V on VANT. But changing the config as you proposed, didn't help.

We've been using this board since late 2017, and we think that it did work in the past. But now, as I've compiled the libnfc-nci again it seems to fail. Could we be running into something like #70 - what do you think?

jgeslin, 2019-07-03:
Hi,

I doesn't looks like issue, because in your side the RF parameters are properly written indicating there is no EEPROM memory corruption.
From my mind the IC seems to be damaged and from the logs I can see this is an engineering PN7150 sample (FW Version: 10.1.10 while release version is 10.01.A0). Where did you get this demoboard ? If it is from distribution, I would suggest you to ask for replacement.

Regards,
Jeremy.

moberhuber, 2019-07-04:
Hello Jeremy,

thanks a lot for your advice! - We have now changed the external power driving VANT from 3.3V to 5V and the card can be read just fine. It is still strange since we think the same hardware config was working in the past, but most important we've got it working now. For newer projects, we've already got chips with 10.1.a0 firmware, so I think we should be good.

From my point of view, the ticket can be closed.

Thanks!
Martin

closed 2019-07-04 by moberhuber