NXPNFCLinux/linux_libnfc-nci #105: "nfcDemoApp poll" is not working in background

saurabhchahan151, 2019-07-03:
Hi,
"nfcDempApp poll" i am able to run and detect the NFC tag, but when i running in the background "nfcDemoApp poll &" its not detecting the NFC tag.

root@localhost:/home/user/nfc/linux_libnfc-nci# ./nfcDemoApp poll &
[1] 11063
root@localhost:/home/user/nfc/linux_libnfc-nci# poll function called...
#########################################################################################

NFC demo
#########################################################################################

Poll mode activated
#########################################################################################
... press enter to quit ...

For working in background is there anything dependent on the source code or command?

jgeslin, 2019-07-05:
Hi,

This is because the application code is expecting any character to quit. When run in background, the application interpret it as a leave request.
To avoid it, simply remove the creation of the ExitThread by commenting out lines 2087 to 2094 in https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/master/demoapp/main.c.

Regards,
Jeremy.

saurabhchahan151, 2019-07-08:
Hi,
It was very helpful. thank you .

closed 2019-07-08 by saurabhchahan151