NXPNFCLinux/linux_libnfc-nci #110: mifare plus tags support

ChevalierDeBalibari, 2019-10-17:
Hello, 

does libnfc_nci support mifare plus and how we can get support if it doesn't?

jgeslin, 2019-10-18:
Hi,

libnfc_nci does support MIFARE plus cards. 
It allows activating the tag, in ISO14443-3A or ISO14443-4 depending of the card security level. Then application can exchange with the card using nfcTag_transceive() API.

You can have a loop to https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/tree/master/tag-raw_example demonstrating use of this API to communicate with ISO14443-3A card. For ISO14443-4, the buffers used as parameters of the API directly convey the Information field of ISO14443-4 I-frames.

However, there is no dedicated implementation to handle MIFARE Plus command set. This must be implemented as part of the application.

Regards,
Jeremy.

ChevalierDeBalibari, 2019-10-20:
Thanks, I will forward this to our application developers. In that case this can be closed.

closed 2019-12-03 by ChevalierDeBalibari