NXPNFCLinux/linux_libnfc-nci #111: Build failure buildroot on Ubuntu 18.10 host

ChevalierDeBalibari, 2019-10-23:
Hello, we hava a problem while building libnfc on ubuntu 18 LTS. We made build script for buildroot and it works on my Manjaro machine, but after transfering to Ubuntu it fails. This is log output produced by buildroot. 

p.s. manual build doesn't work either.  

>>> libnfc_nci R2.4.1 Configuring
cd /home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1 && /home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1/bootstrap
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:11: installing './compile'
configure.ac:6: installing './missing'
Makefile.am: installing './depcomp'
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
**libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:97: error: possibly undefined macro: AC_MSG_ERROR
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:140: error: possibly undefined macro: AC_MSG_NOTICE
autoreconf: /usr/bin/autoconf failed with exit status: 1**
(cd /home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1/ && rm -rf config.cache && PATH="/home/ubuntu/buildroot-at91/output/host/bin:/home/ubuntu/buildroot-at91/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/arm-buildroot-linux-gnueabihf/bin" AR="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ar" AS="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-as" LD="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ld" NM="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-nm" CC="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc" GCC="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc" CPP="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-cpp" CXX="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-g++" FC="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gfortran" F77="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gfortran" RANLIB="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ranlib" READELF="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-readelf" STRIP="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-strip" OBJCOPY="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-objcopy" OBJDUMP="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/usr/bin/gcc" GCC_FOR_BUILD="/usr/bin/gcc" CXX_FOR_BUILD="/usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/home/ubuntu/buildroot-at91/output/host/include" CFLAGS_FOR_BUILD="-O2 -I/home/ubuntu/buildroot-at91/output/host/include" CXXFLAGS_FOR_BUILD="-O2 -I/home/ubuntu/buildroot-at91/output/host/include" LDFLAGS_FOR_BUILD="-L/home/ubuntu/buildroot-at91/output/host/lib -Wl,-rpath,/home/ubuntu/buildroot-at91/output/host/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-as" DEFAULT_LINKER="/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ld" CPPFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  " LDFLAGS="" FCFLAGS=" -Os " FFLAGS=" -Os " PKG_CONFIG="/home/ubuntu/buildroot-at91/output/host/bin/pkg-config" STAGING_DIR="/home/ubuntu/buildroot-at91/output/host/arm-buildroot-linux-gnueabihf/sysroot" INTLTOOL_PERL=/usr/bin/perl ac_cv_lbl_unaligned_fail=yes ac_cv_func_mmap_fixed_mapped=yes ac_cv_func_memcmp_working=yes ac_cv_have_decl_malloc=yes gl_cv_func_malloc_0_nonnull=yes ac_cv_func_malloc_0_nonnull=yes ac_cv_func_calloc_0_nonnull=yes ac_cv_func_realloc_0_nonnull=yes lt_cv_sys_lib_search_path_spec="" ac_cv_c_bigendian=no   CONFIG_SITE=/dev/null ./configure --target=arm-buildroot-linux-gnueabihf --host=arm-buildroot-linux-gnueabihf --build=x86_64-pc-linux-gnu --prefix=/usr --exec-prefix=/usr --sysconfdir=/etc --localstatedir=/var --program-prefix="" --disable-gtk-doc --disable-gtk-doc-html --disable-doc --disable-docs --disable-documentation --with-xmlto=no --with-fop=no --disable-dependency-tracking --enable-ipv6 --disable-nls --disable-static --enable-shared   )
configure: WARNING: unrecognized options: --disable-gtk-doc, --disable-gtk-doc-html, --disable-doc, --disable-docs, --disable-documentation, --with-xmlto, --with-fop, --enable-ipv6, --disable-nls
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/bin/bash: /home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1/missing: No such file or directory
configure: WARNING: 'missing' script is too old or missing
checking for arm-buildroot-linux-gnueabihf-strip... /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-g++ accepts -g... yes
checking for style of include used by make... GNU
checking dependency style of /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-g++... none
checking for arm-buildroot-linux-gnueabihf-gcc... /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc
checking whether we are using the GNU C compiler... yes
checking whether /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc accepts -g... yes
checking for /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc option to accept ISO C89... none needed
checking whether /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc understands -c and -o together... yes
checking dependency style of /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc... none
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-buildroot-linux-gnueabihf
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-gcc... /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ld
checking if the linker (/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-nm
checking the name lister (/home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to arm-buildroot-linux-gnueabihf format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-ld option to reload object files... -r
checking for arm-buildroot-linux-gnueabihf-objdump... /home/ubuntu/buildroot-at91/output/host/bin/arm-buildroot-linux-gnueabihf-objdump
checking how to recognize dependent libraries... pass_all
[... cut ...]
CXXFLAGS=-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os  
./configure: line 16664: syntax error near unexpected token `UDEV,'
./configure: line 16664: ` if PKG_CHECK_MODULES(UDEV, libudev,'
package/pkg-generic.mk:228: recipe for target '/home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1/.stamp_configured' failed
make[1]: *** [/home/ubuntu/buildroot-at91/output/build/libnfc_nci-R2.4.1/.stamp_configured] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2

iprotonotarios, 2019-12-17:
You probably need to install pkg-config.

ChevalierDeBalibari, 2020-12-14:
This is resolved. Some build dependencies like autotools were missing.

closed 2020-12-14 by ChevalierDeBalibari