NXPNFCLinux/linux_libnfc-nci #112: I can't find some API reference in the "Linux_NFC_API_Guide.html",Is there any other document for use?

niguang1, 2019-11-07:
When i run the demoAPP in write mode and put a TypeA card on the reader,it Cyclic display information below:
Waiting for a Tag/Device...                                                                                     
        NFC Tag Found                                                                                                                           
        Type :         'Type A - Mifare Classic'                                
        NFCID1 :        'B9 D9 E8 78 '                                          
                NDEF Content : NO, mode=3, tech=8                                                                                             
        Formating tag to NDEF prior to write ...                                
        Tag formating failed                                                    
        Write Tag Failed                                                        
        NFC Tag Lost                                                                             
Waiting for a Tag/Device... 
So i back to check my code,but i can't find an introduction to below:
if(nfcTag_isFormatable(TagInfo.handle))
if(nfcTag_formatTag(TagInfo.handle) == 0x00)
Where i can get complete documentation?Could you give me somd advice?

jgeslin, 2019-11-07:
The type A card is detected as MIFARE Classic card and not NDEF formatted yet.
Then the demo application tries to format it using MIFARE factory default keys. Since it fails it probably means the card have already been updated with different MIFARE keys and the demo application cannot knows them, so cannot format.

What are you exactly trying to do ?

niguang1, 2019-11-07:
What you mean is when i put a empty card on the reader, it can be formatted and i can write a NDEF data on it ?

niguang1, 2019-11-07:
In fact, I would prefer a complete reference document.If accessible,Thanks very much.

jgeslin, 2019-11-07:
Unfortunately, there is no other document than https://www.nxp.com/docs/en/application-note/AN11697.pdf and https://github.com/NXPNFCLinux/linux_libnfc-nci/tree/master/doc.

niguang1, 2019-11-07:
OK,thanks very much.

closed 2019-11-07 by niguang1