NXPNFCLinux/linux_libnfc-nci #114: What is the maximum transmission speed of PN7150 in P2P mode by libnfc-nci library?

niguang1, 2019-11-13:
I set one device in send mode and another in receive mode,and send a 13112 bytes file to the receiver.it costs about 8 seconds finally.
I know that NFC's max speed is 424kbps,but what is the max speed of PN7150?
Is this useful to change IIC's speed bigger? How to improve the transmission speed of PN7150’s IIC?
And is there any way to improve the transmission speed of PN7150?

jgeslin, 2019-11-18:
Indeed, the SNEP protocol add overhead to the communication, and part of the time is also spent in the processing of packets from both sides of the communication.
However, what I2C speed are you using ?

jgeslin, 2019-12-13:
I did a quick test using 2 raspberry pi (1 RPI 3B+ and 1 RPI B+).
Exchanging one NDEF message of 13140 bytes from one side takes approx. 2500ms.
When updating the I2C speed of both boards to 400kbps (instead of 100kbps by default) I get a transfer time of approx. 650ms.

See attached updated p2p example files (one per side) for the testing.
[p2p_examples.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/3961042/p2p_examples.zip)

closed 2019-12-20 by niguang1