NXPNFCLinux/linux_libnfc-nci #115: Multiple controllers in the same I2C channel

lucar91, 2019-11-14:
Hi, 
I connected 2 NFC controller in the same embedded device, and I have two different i2c_address (0x29 and 0x2A). If I switch the connected controller,  then I work with one controller at a time, and I change the LD_LIBRARY_PATH before starting the application, the application works  (LD_LIBRARY_PATH=/usr/local/lib29 or LD_LIBRARY_PATH=/usr/local/lib2A. The libraries are builds with the correct configuration of the address and the different GPIOs for each controller).

I should like both controllers to work together in the same application, but this means that I have to change LD_LIBRARY_PATH at runtime, and this is impossible.

A solution could be to create two applications: the read_and_write application that works with the linux_nfc_api, and the main application.

The main application calls a sh script that contains the new LD_LIBRARY_PATH and the first application. Example:
1)script_address_29:
LD_LIBRARY_PATH=...29
read_and_write

2)script_address_2A:
LD_LIBRARY_PATH=...2A
read_and_write

If we need to run the first controller, we call the script one, otherwise, the script two.

 There are other solutions without the use of two applications?

jgeslin, 2019-11-18:
The solution to your problem could be to build 2 applications App29 and App2A statically linked to the related library instead of dynamic loading.
Please have a look to this https://www.mpcdf.mpg.de/services/computing/software/libraries/static-and-dynamic-linking-on-linux-systems in order to see how to update Makefile accordingly.

lucar91, 2019-11-22:
Firsts of all, thanks for your reply. 
My problem is this: whit your solution, I would have three processes. The main process will call the other two processes (2a  and 29 apps) alternately.    

Your solution (similar to mine): 

int main(){
	system ("sh process_2A");
        ...
	system ("sh process_29");
        ...
}

I would like to work with only one application, the main process. 

Example:
main(){
	function_of_nfc_with_lib29();
        ...
	function_of_nfc_with_lib2A();
        ...
}

jgeslin, 2019-11-22:
Then I don't see any clever solution else than doing a big rework of the libnfc-nci to instantiate to handle with different I2C addresses.
The libnfc-nci was indeed not designed for such a thing.

lucar91, 2019-11-25:
Perfect, that's what I wanted to know. Thanks so much

closed 2019-11-29 by lucar91

lucar91, 2020-01-30:
Hi, in case of multiple controllers with the procedure described above (app1 that call app2 to read NFC tag), can I think to not call  "nfcManager_doDeinitialize()" function?

**i.e.** 
app1_main{
   while(){
    app2 _controller1
    app2_controller2
    sleep 1;
   }
}
**Actually:**

app2_controller1{
//some operations
   ...
   nfcManager_doDeinitialize();
   return;
}
app2_controller2{
   //some operations
   ...
   nfcManager_doDeinitialize();
   return;
}

**My idea:**
app2_controller1{
   //some operations
   ...
   return;
}
app2_controller2{
  //some operations
  ...
  return;
}

My solution works, but what does it mean not to call the  "nfcManager_doDeinitialize()" function?

jgeslin, 2020-02-03:
Hi,

The "nfcManager_doDeinitialize()" function is properly stopping NFC functionality (de-allocate all memory and threads) and deactivates NFC Controller (put it to hard power down mode to limit the power consumption when). Thus it is recommended to call it when NFC function is not used anymore (at least for a certain amount of time).
However, if your device expect to keep NFC running all the time (just stopped when shutdown) then there is no need to use this API.

Regards,
Jeremy.