NXPNFCLinux/linux_libnfc-nci #116: ISO_DEP_MAX_TRANSCEIVE support

alci63, 2019-11-28:
Hi,

I have a problem where my mobile phone (Samsung Galaxy S9 Plus, presumably including NXP PN80T NFC controller), running Lineageos based /e/ OS, fails to decrypt PGP encrypted messages using OpenKeyChain and a Yubikey 5 NFC.

The error message says `error transceive length exceeds supported maximum` 

This [commit review](https://android-review.googlesource.com/q/status:merged+branch:master+topic:max_transceive_length ) suggests that in Android, the ISO_DEP APDU max length can be set using the ISO_DEP_MAX_TRANSCEIVE tag in libnfc conf files, or else will default to 261 bytes.

Is this ISO_DEP_MAX_TRANSCEIVE tag supported within this library ? I couldn't find the tag in the example conf files. Or is there another way to enable larger APDUs ?

Thanks in advance

jgeslin, 2019-12-06:
Hi,

I do not understand what you are exactly looking for but I can confirm linux_libnfc-nci stack do not support ISO_DEP_MAX_TRANSCEIVE parameter from configuration file (it will simply be ignored).

Regards,
Jeremy.

alci63, 2019-12-06:
Thanks for the information.
The purpose is to handle some NFC transactions with APDU longer than the default 261 bytes size.
This is the case for example with yubikey openpgp app communicating with openkeychain.

jgeslin, 2019-12-12:
The Linux_libnfc-nci API nfcTag_transceive allows sending/receiving more than 256 bytes, triggering use of extended APDU (RF chaining) while in reader mode. Is that what you are looking for ?

alci63, 2019-12-12:
I'm not quite sure. The relevant code that fails in my use case is [here in Openkeychain's github](https://github.com/open-keychain/open-keychain/blob/master/OpenKeychain/src/main/java/org/sufficientlysecure/keychain/securitytoken/OpenPgpCommandApduFactory.java)

This fails with `error transceive length exceeds supported maximum`. I think this is due to an error below in the software stack, and I am looking for a way to set this ISO_DEP_MAX_TRANSCEIVE in the NFC controller driver of my device, under Lineageos... Not sure it is really clear... in fact I'm struggling to understand how this works...

jgeslin, 2019-12-13:
Looks your question doesn't relate to Linux_libnfc-nci stack at all but rather about the Android AOSP NFC service. The ISO_DEP_MAX_TRANSCEIVE configuration then relates to the phone configuration. To change it you'll probably need to root the phone (if not already done), then to find the related configuration file (depending of the Android version,for instance on Android P version it should be something similar to /vendor/etc/libnfc-nci.conf.

See here similar instruction for other phone: https://forum.xda-developers.com/nexus-5x/orig-development/rom-lineageos-15-1-nexus-5x-bullhead-t3724575/page68

alci63, 2019-12-13:
Thanks a lot. You are absolutely right, and I managed to get my phone (S9+ under /e/ OS) working with my Yubikey 5 NFC, by editing /vendor/etc/libnfc-brcm.conf on my phone (adding ISO_DEP_MAX_TRANSCEIVE_LENGTH=0xFEFF).

Sorry for the noise, and thanks for being helpful !

closed 2019-12-13 by alci63

nh2, 2020-12-12:
Related: https://github.com/open-keychain/open-keychain/issues/2501