NXPNFCLinux/linux_libnfc-nci #117: problem getting PN7150 into low power mode

ChevalierDeBalibari, 2019-12-02:
Hello, 

we have a problem getting PN1750 into low power mode. 

libnfc-nxp-pn547.conf looks like it is already configured for LP mode but if i change any parameter here nothing happens.
libnfc-nxp-init.conf doesn't have enabled detect mode:

NXP_NFC_PROFILE_EXTN={20, 02, 05, 01, A0, 44, 01, 00}

however, if i manually put configuration as provided in AN11697 library fails to load.

jgeslin, 2019-12-02:
Hi,

Are you refering to the PN7150 standby mode or to the Low power discovery mode (as described in https://www.nxp.com/docs/en/application-note/AN11757.pdf) ?

ChevalierDeBalibari, 2019-12-03:
Low power discovery mode. We measured pulses on antenna, and they are generated every ~200ms.

jgeslin, 2019-12-03:
Low power mode discovery for PN7150 is configured thanks to NXP_CORE_CONF_EXTN parameter defined in **libnfc-nxp-pn548.conf** file. Indeed, libnfc-nxp-pn547.conf relates to PN7120 NFC Controller.
Default configuration for PN7150 in NXP_CORE_CONF_EXTN is Low power discovery mode "disabled".
For detailed description of PN7150 Low power discovery mode configuration please refer to https://www.nxp.com/docs/en/application-note/AN11757.pdf.

ChevalierDeBalibari, 2019-12-03:
Thnaks! I will check it right now!

ChevalierDeBalibari, 2019-12-03:
Now there isn't any error reported. How can I check that system is working? 

I tried with nfcDemoApp but it doesn't detect tag. Can we do it with it or through libnfc-nci API?

jgeslin, 2019-12-03:
You can able libnfc-nci traces setting parameters NXPLOG_xxx_LOGLEVEL to 0x03 in libnfc-nxp-init.conf.

ChevalierDeBalibari, 2019-12-03:
Already did it. Here is the log:

`#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)

NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)

NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
NxpHal:     Entering phNxpNciHal_init_monitor
NxpHal:     Returning with SUCCESS
NxpExtns:   find found NXP_NFC_DEV_NODE=/dev/pn544

NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE=/dev/pn544

NxpTml:     phTmlNfc_i2c_open_and_configure

NxpTml:     Opening port=/dev/pn544

NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpFunc:    Nfc: NFC_SetTraceLevel () new_level = 0

NxpTml:     phTmlNfc_i2c_reset(), VEN level 0
NxpTml:     phTmlNfc_i2c_reset(), VEN level 1
NxpTml:     PN54X - Tml Reader Thread Started................

NxpTml:     PN54X - Read requested.....

NxpTml:     PN54X - Invoking I2C Read.....

NxpTml:     PN54X - Tml Writer Thread Started................

NxpTml:     PN54X - Tml Writer Thread Running................

NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =   4 > 20000101
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     thread started
NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   6 > 400003001101
NxpTml:     PN54X - Posting read message.....
[... 426 lines cut ...]
NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................

NxpHal:     write successful status = 0x0
NxpHal:     Response timer started
NxpHal:     Waiting after ext cmd sent
NxpTml:     PN54X - I2C Read successful.....

NxpNciR:     len =   4 > 40020105
NxpTml:     PN54X - Posting read message.....

NxpHal:     read successful status = 0x0
NxpHal:     phNxpNciHal_print_res_status: response status =STATUS_SYNTAX_ERROR
NxpHal:     Invalid Data from config file . Aborting..
NxpTml:     PN54X - Write requested.....

NxpTml:     PN54X - Invoking I2C Write.....

NxpNciX:     len =  10 > 21030703800181018201
NxpTml:     PN54X - I2C Write successful.....

NxpTml:     PN54X - Posting Fresh Write message.....

NxpTml:     PN54X - Tml Writer Thread Running................`

ChevalierDeBalibari, 2019-12-03:
Still, no tag is detected.

jgeslin, 2019-12-03:
You wrongly updated parameter NXP_CORE_CONF_EXTN parameter. Indeed, third byte relates to the complete frame size and fourth one to the number of parameter set in this command.
Here is the correct frame definition according to the configuration you seem to expect:
NXP_CORE_CONF_EXTN={20, 02, 11, 04,
    A0, 5E, 01, 01,
    A0, 40, 01, 00,
    A0, 41, 01, 04,
    A0, 43, 01, 00
}

ChevalierDeBalibari, 2019-12-03:
thanks! it works now. sensitivity is not great but we will fix it up

closed 2019-12-03 by ChevalierDeBalibari