NXPNFCLinux/linux_libnfc-nci #118: Is there anyway to set PN7150 at both P2P and HCE mode?

niguang1, 2019-12-16:
The description of function nfcManager_enableDiscovery says that :

reader_only_mode      indicates if enable reader only mode. (Means no P2P or HCE)

Is this means that if i set reader_only_mode as 1 ,only one of P2P and HCE mode can be setted;and if i set reader_only_mode as 0,both mode can be setted?

Now i have finished three mode of PN7150 separated,but  have trouble on Integration of these three modes,can you give me some advice ?

jgeslin, 2019-12-17:
Basically, when you present a device supporting P2P and HCE in front of a device supporting Reader and P2P (both based on Linux libnfc-nci stack), the devices will always discover the counterpart in P2P mode (this mode been preferred) unless you select "reader_only_mode" option on the reader and P2P device side (in this case devices will detect in reader/HCE mode).

niguang1, 2019-12-20:
Finally i wrote two programs to solve this problem,because I know when i will start once send,hopping to  run well.
Thank you for your help. My project is coming to an end.

closed 2019-12-20 by niguang1