NXPNFCLinux/linux_libnfc-nci #120: Further details about the supported NXP NFC controller

lorn10, 2020-06-02:
Hi there NXPNFCLinux devs

This is not a bug report. I just want to note here, that the documentation regarding the effectively supported NXP NFC controllers could be somewhat improved.

For example, what means "NCI based NXP NFC Controllers" in detail? There are **a lot** of different NXP NFC controllers avaiable. It is quite unclear which one are supported by this stack and which aren't. It looks that this drivers stack is developed primary for newer generation NXP NFC controller models.

There should be at least a remark regarding any unsupported NXP NFC controller and / or controller firmware. According to the information I found, it looks that some older NXP controllers like the quite old NXP PN544 chip family are mostly unsupported. This is true for the PN544 C1 variant, the support of the PN544 C2 (and newer) may vary according the chip firmware level.

  -  PN544 C1, PN544 C2, PN544 C3
  -  PN65N (PN544 C2 + SmartMX secure element)
  -  PN65O (PN544 C3 + SmartMX secure element)

Can someone confirm this? 

However, for those old NFC controllers, there exist as an the Linux PN544 Kernel driver. But also for that alternative driver it is not really clear which PN544 chip incarnation is effectively supported. ;-)

Kind regards

jgeslin, 2020-06-11:
Hi,

This software stack supports NCI based NFC Controllers from NXP, the most obvious ones being PN7150 and PN7120.
PN544 family (and derivates PN65N and PN65O) offers HCI interface (not NCI) thus are not supported by this software stack. Same statement for PN532/PN533 offering TAMA protocol interface.

Thanks for the remark, the README file has been updated accordingly to reflect supported controllers references.

Regards,
Jeremy.

lorn10, 2020-06-11:
Hello and thanks Jeremy for the clarification.

This is exactly the answer which I was looking for. So PN544 (and older EOL) NFC controllers are supported only through the alternative opensource Linux PN544 Kernel driver. ;-)

Regards
Clemens

closed 2020-06-11 by lorn10