NXPNFCLinux/linux_libnfc-nci #121: Reading gets stuck in function nfcTag_isNdef()

Hansi1990, 2020-08-13:
I implemented the current version of your library on my arm target. 
During initialization of my application, I start a thread that runs the polling mode using the _cmd_poll()_ function, same as in the demo code.  At first I thought everything (also writing) seemed to work fine, but then I was stressing the system a little bit by placing and removing the NFC tag many times really fast. After around 10 successful reads, the library gets stuck and blocks my thread.

However, running the nfcDemoApp seems to work fine, no freeze detected.

Can you give me a hint where I could start debugging? In my understanding, the library shouldn't be allowed to block there at all.

```
[... 10 successful reads...]

Waiting for a Tag/Device...

1970:01:01-01:53:49.509 NxpTml:     PN54X - I2C Read successful.....
1970:01:01-01:53:49.509 NxpNciR:     len =  26 > 61051701010200FF010C44000704BC6312436280010000000000
1970:01:01-01:53:49.509 NxpTml:     PN54X - Posting read message.....
1970:01:01-01:53:49.510 NxpHal:     read successful status = 0x0
1970:01:01-01:53:49.510 NxpHal:     NxpNci: RF Interface = Frame RF
1970:01:01-01:53:49.510 NxpHal:     NxpNci: Protocol = T2T
1970:01:01-01:53:49.510 NxpHal:     NxpNci: Mode = A Passive Poll
1970:01:01-01:53:49.510 NxpFunc:    NfcAdaptation::HalDeviceContextDataCallback: len=26
1970:01:01-01:53:49.510 NxpFunc:    Nfc: Process Event mt:3
1970:01:01-01:53:49.510 NxpFunc:    Nfc: NFC received ntf gid:1
1970:01:01-01:53:49.510 NxpFunc:    Nfc: P2P_Prio_Logic
1970:01:01-01:53:49.510 NxpFunc:    Nfc: nfa_dm_p2p_prio_logic type = 0x1
1970:01:01-01:53:49.511 NxpFunc:    Nfc: returning TRUE
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfc_set_state 4->5
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfc_ncif_proc_activate:23 0, mode:0x00
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_disc_discovery_cback (): event:0x4004
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_disc_sm_execute(): state: 1, event:5 disc_flags: 0x1
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_disc_new_state(): old_state: 1, new_state: 4 disc_flags: 0x1
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_disc_notify_activation (): tech_n_mode:0x0, proto:0x2
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_disc_get_disc_mask (): tech_n_mode:0x0, protocol:0x2, disc_mask:0x2
1970:01:01-01:53:49.511 NxpFunc:    Nfc: activated_protocol:0x2, activated_handle: 0x1
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_dm_poll_disc_cback (): event:0x01
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_rw_handle_event event: 0x601, flags: 00000041
1970:01:01-01:53:49.511 NxpFunc:    Nfc: nfa_rw_activate_ntf
1970:01:01-01:53:49.512 NxpFunc:    Nfc: RW_SetActivatedTagType protocol:2, technology:0, SAK:0
1970:01:01-01:53:49.512 NxpFunc:    Nfc: nfa_dm_notify_activation_status (): status:0x0
1970:01:01-01:53:49.512 NxpFunc:    nfaConnectionCallback: event= 5
1970:01:01-01:53:49.512 NxpFunc:    checkforTranscation: enter; event=0x5 transaction_data.current_transcation_state = 0x0
1970:01:01-01:53:49.512 NxpFunc:    other event clearing flag
1970:01:01-01:53:49.512 NxpFunc:    checkforTranscation: exit; event=0x5 transaction_data.current_transcation_state = 0x0
1970:01:01-01:53:49.512 NxpFunc:    nfaConnectionCallback: NFA_ACTIVATED_EVT: gIsSelectingRfInterface=0, sIsDisabling=0
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::setActivationState: state=2
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::IsSameKovio: enter
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::discoverTechnologies (activation): enter
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::processNotification (Activation): RF disc_id=01; protocol=02, mNumDiscList=0
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::processNotification; mNumTechList=2
1970:01:01-01:53:49.512 NxpFunc:    NfcTag::discoverTechnologies (activation): index=0; tech=1; handle=1; nfc type=2
1970:01:01-01:53:49.513 NxpFunc:    NfcTag::discoverTechnologies (activation): index=1; tech=9; handle=1; nfc type=2
1970:01:01-01:53:49.513 NxpFunc:    NfcTag::discoverTechnologies (activation): exit
1970:01:01-01:53:49.513 NxpFunc:    NfcTag::createNativeNfcTag: enter
1970:01:01-01:53:49.513 NxpFunc:    createNativeNfcTag: Selected: index=1; tech=9; handle=1; nfc type=2
1970:01:01-01:53:49.513 NxpFunc:    NfcTag::setNfcTagUid: tech A
1970:01:01-01:53:49.513 NxpFunc:    nativeNfcTag_onTagArrival: notify tag is ready
        NFC Tag Found

unlock() failed No such file or directory
1970:01:01-01:53:49.513 NxpFunc:    Unable to create the thread
1970:01:01-01:53:49.513 NxpFunc:    Nfc: NFA_Deactivate (): sleep_mode:0
1970:01:01-01:53:49.514 NxpTml:     PN54X - Read requested.....
1970:01:01-01:53:49.514 NxpTml:     PN54X - Invoking I2C Read.....
unlock() failed No such file or directory
        Type :         'Type A - Mifare Ul'
        NFCID1 :        '04 BC 63 12 43 62 80 '
1970:01:01-01:53:49.520 NxpFunc:    nativeNfcTag_checkNdef: enter; handle=1
1970:01:01-01:53:49.522 NxpFunc:    nativeNfcTag_checkNdef: try NFA_RwDetectNDef
1970:01:01-01:53:49.524 NxpFunc:    nativeNfcTag_checkNdef: NfcTag::getInstance ().mTechLibNfcTypes[1]=2
1970:01:01-01:53:49.526 NxpFunc:    Nfc: NFA_RwDetectNDef

```

jgeslin, 2020-08-13:
Hi,

Trace: 

> 1970:01:01-01:53:49.513 NxpFunc:    Unable to create the thread

shows an issue while creating a thread inside the library (for continuous tag presence check purpose). 
From my mind you have a memory leak in your application leading to this issue, explaining why this is not observed with the demo app.
I agree that the library should be robust against such thread creation issue, but solving the memory lead should resolve the issue.

Regards,
Jeremy.

Hansi1990, 2020-08-13:
Hi Jeremy,
Thanks for the hint. I'll look into this and keep you posted.

Hansi1990, 2020-08-14:
Hi Jeremy,
you were right. It wasn't a memory leak, but every once in a while my system was at the limit of how many new threads my system was able to handle. In some situations, it was not able to create a new thread with pthread_create() and it fails. Now I made sure that new threads can be created, so it runs stable.

Maybe it can be discussed in the future to have a return value indicating an error instead of getting stuck in the function as a robust measure to be able to handle that specific error in the function.

Thanks for your help!

Tohkai, 2021-04-01:
> Hi Jeremy,
> you were right. It wasn't a memory leak, but every once in a while my system was at the limit of how many new threads can be handled. In some situations, it was not able to create a new thread with pthread_create() and it fails. Now I made sure that new threads can be created, so it runs stable.
> 
> Maybe it can be discussed in the future to have a return value indicating an error instead of getting stuck in the function as a robust measure to be able to handle that specific error in the function.
> 
> Thanks for your help!

Hello. Can you tell me how you "made sure that new threads can be created" ?  I am having similar problems, and you didn't give us many leads by saying this. 
Thanks!