NXPNFCLinux/linux_libnfc-nci #122: `select` timeout treated as IO error

hefroy, 2020-10-09:
The following line calls `select` with a timeout:

https://github.com/NXPNFCLinux/linux_libnfc-nci/blob/2aff6448cf6a86897d3d1bacb0a0943a594aa9ce/src/halimpl/pn54x/tml/i2c/phTmlNfc_i2c.c#L367

However if the timeout occurs, -1 is returned. This is treated by the calling code in the same way as an IO error, and after several attempts the dreaded `CUSTOM> too many i2c read errors: ABORTING...` error occurs.

The library currently works as the `poll` file operation of the pn5xx_i2c driver is not yet implemented in the Linux kernel.
Adding support for it in the kernel would allow other drivers to use the pn5xx device in non-blocking mode. (Kernel patch not yet submitted.)

jgeslin, 2020-10-12:
if the select fails with timeout it means the pn5xx device is not available, and the library cannot run without connection to NFC Controller since it doesn't make any sense.

Regarding your proposal about sharing pn5xx device with other user-space application, it must be avoided. Indeed, the library must keep trace of the NFC Controller state to properly operate. If another application impact the NFC controller state it may end-up in NFC deadlock.

hefroy, 2020-10-12:
> if the select fails with timeout it means the pn5xx device is not available
This isn't true. If `select` returns with a timeout, it means that the device was not readable within the given time. I.e. the IRQ line of the device remained low for the duration of the `select` call. This will occur once the device is in discovery mode and waiting for arrival of a tag.
The existing libnfc-nci code 'works' because the pn5xx_i2c driver does not _currently_ implement the `poll` file operation, hence calling `select` returns that the device is _always_ readable. The library then uses a blocking `read` operation to wait for the next data to become available (which could be never).

> Regarding your proposal about sharing pn5xx device with other user-space application, it must be avoided.
I'm not proposing to share the pn5xx_i2c device with the libnfc-nci library, indeed I'm not planning to use the library at-all. Instead I'm using my own user-space code with the pn5xx_i2c driver. I'm thinking of submitting a kernel patch that implements the `poll` operation so that the pn5xx_i2c driver, so that others can do the same, hence I'm letting you know as a courtesy.

In case you're interested, here's my current implementation of the `poll` file operation, note the temporary hack to allow the driver to continue to work with libnfc-nci in blocking mode. (My code is using the code in non-blocking mode):
```
static unsigned int pn54x_dev_poll(
	struct file *filp,
	struct poll_table_struct *wait)
{
	struct pn54x_dev *pn54x_dev = container_of(filp->private_data, struct pn54x_dev, pn54x_device);
	unsigned int events = 0;

	// FIXME: Temporary hack to work with libnfc-nci which performs `select` with a timeout, but
	//        when the timeout occurs, it treats it as an IO error.
	if ((filp->f_flags & O_NONBLOCK) == 0) {
		pr_debug("%s : fake readble", __func__);
		return POLLIN | POLLRDNORM;
	}

	poll_wait(filp, &pn54x_dev->read_wq, wait);
	if (!gpio_get_value(pn54x_dev->irq_gpio)) {
		pr_debug("%s : not readble", __func__);
		pn54x_enable_irq(pn54x_dev);
	} else {
		pr_debug("%s : readble", __func__);
		events = POLLIN | POLLRDNORM;
	}
	return events;
}
```

hefroy, 2020-10-13:
I've just realized the pn5xx_i2c driver is not part of the mainline kernel, my mistake!

closed 2020-10-13 by jgeslin