NXPNFCLinux/linux_libnfc-nci #123: Run "Make"  failed in Fedora33

HowardKL, 2021-03-19:
When I run "make" command in Fedora33, I met the following error.

- Collect2 error: ld returned 1 exit status

This error might be made from GCC10 by referring to the following issue.
[https://github.com/intel/intel-hybrid-driver/issues/27](url)

Could someone help to fix the code to compel with GCC10 rule?
Thank.

jgeslin, 2021-03-19:
Hi,

Can you provided detailled error logs ? 
The current information is not enough to analyze the issue faced.

Jeremy.

HowardKL, 2021-03-19:
I'd appreciate for your quickly reply @jgeslin .
Please see the attached file.[log.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/6169532/log.txt)
Thanks.
`/usr/bin/ld: src/halimpl/pn54x/tml/i2c/.libs/phTmlNfc.o:(.bss+0x8): multiple definition of 'fragmentation_enabled'; src/halimpl/pn54x/tml/i2c/.libs/phTmlNfc_i2c.o:(.bss+0x0): first defined here
/usr/bin/ld: src/service/extns/src/.libs/phNxpExtns.o:(.bss+0x0): multiple definition of 'gphNxpExtns_Context'; src/service/extns/src/mifare/.libs/phNxpExtns_MifareStd.o:(.bss+0x160): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:2028: libnfc_nci_linux.la] Error 1
make[1]: Leaving directory '/home/fedora/Desktop/linux_libnfc-nci'
make: *** [Makefile:1208: all] Error 2`

jgeslin, 2021-03-22:
Please try attached patch.
[aarch64-unknown-linux-musl_toolchain.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/6183101/aarch64-unknown-linux-musl_toolchain.zip)

HowardKL, 2021-03-26:
Thanks for your help @jgeslin .
This error could be fixed by the patch.

closed 2021-03-26 by HowardKL