NXPNFCLinux/linux_libnfc-nci #124: Null ptr deref wh

corneliusclaussen, 2021-04-15:
#0  0x00000000 in ?? ()
#1  0xb6f5e2e4 in Mfc_CheckNdef_Completion_Routine (NdefCtxt=0x0, status=255) at src/service/extns/src/mifare/phNxpExtns_MifareStd.c:394
#2  0xb6f65d70 in phFriNfc_MifStd_H_Complete (NdefMap=0x2e9b8, Result=255) at src/service/extns/src/mifare/phFriNfc_MifareStdMap.c:1367
#3  0xb6f6532c in phFriNfc_MifareStdMap_Process (Context=0x2e9b8, Status=255) at src/service/extns/src/mifare/phFriNfc_MifareStdMap.c:1020
#4  0xb6f605a4 in Mfc_RecvPacket (buff=0x32e25 "@\003", buffSz=2 '\002') at src/service/extns/src/mifare/phNxpExtns_MifareStd.c:1495
#5  0xb6f6dc6c in phNxpExtns_ProcessSysMessage (msg=0xb50fcc00) at src/service/extns/src/phNxpExtns.c:472
#6  0xb6f6de70 in phNxpExtns_SendMsg (sysmsg=0xb50fcc00) at src/service/extns/src/phNxpExtns.c:540
#7  0xb6f6d4a8 in EXTNS_MfcCallBack (buf=0x32e25 "@\003", buflen=2) at src/service/extns/src/phNxpExtns.c:131
#8  0xb6f4317c in nativeNfcTag_doTransceiveStatus (status=0 '\000', buf=0x32e25 "@\003", bufLen=2) at src/service/interface/nativeNfcTag.cpp:917
#9  0xb6f51ce0 in nfaConnectionCallback (connEvent=9 '\t', eventData=0xb50fccc0) at src/service/interface/nativeNfcManager.cpp:634
#10 0xb6ebb0ec in nfa_dm_conn_cback_event_notify (event=9 '\t', p_data=0xb50fccc0) at src/libnfc-nci/nfa/dm/nfa_dm_act.c:676
#11 0xb6e8ccb8 in nfa_rw_raw_mode_data_cback (conn_id=0 '\000', event=24579, p_data=0xb50fcd88) at src/libnfc-nci/nfa/rw/nfa_rw_act.c:2675
#12 0xb6ee4884 in nfc_data_event (p_cb=0xb6fb2b58 <nfc_cb+4>) at src/libnfc-nci/nfc/nfc/nfc_ncif.c:2013
#13 0xb6ee4d2c in nfc_ncif_proc_data (p_msg=0x32e10) at src/libnfc-nci/nfc/nfc/nfc_ncif.c:2136
#14 0xb6ee1be8 in nfc_ncif_process_event (p_msg=0x32e10) at src/libnfc-nci/nfc/nfc/nfc_ncif.c:689
#15 0xb6eddc34 in nfc_task (param=0) at src/libnfc-nci/nfc/nfc/nfc_task.c:440
#16 0xb6ece144 in gki_task_entry (params=3069910760) at src/libnfc-nci/gki/ulinux/gki_ulinux.c:132
#17 0xb6e3d300 in start_thread (arg=0xb50fd450) at pthread_create.c:477
#18 0xb6dc1208 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6

corneliusclaussen, 2021-04-15:
Hi, I get the above stack trace when presenting a mifare classic card to the demo app. It crashes when it dereferences the callback ptr: 
(*gphNxpExtns_MifareStd_Context.p_conn_cback) (NFA_NDEF_DETECT_EVT, &conn_evt_data);

in STATIC void Mfc_CheckNdef_Completion_Routine(void *NdefCtxt, NFCSTATUS status). I cannot find that .p_conn_cback ever gets initialized...

jgeslin, 2021-04-16:
Many thanks for the fix.
It is now integrated.

closed 2021-04-16 by jgeslin