NXPNFCLinux/linux_libnfc-nci #127: PN7150 Type 2 tag emulation

antznin, 2021-04-26:
Hi,

I'm currently working with this library and the PN7150, where I see that it is possible to emulate a Type 4 tag. However, I would like to emulate a Type 2 tag. Is it possible ?

I haven't found any information in the PN7150 User Manual (only ISODEP and T3T mentionned), nor in this repository.

Could the type 2 protocol be implemented like the type 4 is ? That is, analyze the reader's data received and respond using `nfcHce_sendCommand`.

Any tips or links to examples for other devices would be really appreciated.

Regards,
Antonin

jgeslin, 2021-04-26:
Hi,

PN7150 supports card emulation mode only as T4T and/or T3T. Thus this library only provides API for those modes and there is no example for T2T card emulation.

Regards,
Jeremy.

antznin, 2021-04-26:
Thank you for the quick answer.

Regards,
Antonin

closed 2021-04-26 by antznin