NXPNFCLinux/linux_libnfc-nci #130: Type B card emulation

matthew5025, 2021-06-06:
I'm using this library with the PN7150, which should be capable of Type B HCE, but it does not seem to work.
Is there some additional configuration I need to do to enable it?

jgeslin, 2021-06-07:
Hi,

The Linux libnfc-nci library running on top of PN7150 NFC controller supports Type B HCE.
You can refer to the demo application in "share" mode or also to the specific card_emulation example: https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/tree/master/card-emulation_example
Pay attention that the Type B RF communication is more difficult to obtain than Type A and requires performant Type B reader.

From my side I'm capable to perform Type B HCE scenario using PN7150 demokit and Raspberry Pi setup, with Samsung Galaxy S9 as reader counterpart (requiring to disable HCE Type A thanks to HOST_LISTEN_TECH_MASK parameter within libnfc-nci.conf file otherwise communication is always established in this mode). But obviously it requires close and dedicated placement of the phone against the PN7150 board antenna to achive this.

Regards,
Jeremy.

matthew5025, 2021-06-07:
Hi Jeremy, 

Thanks for the advice! Utilising a more powerful reader did the trick.

On the part of the mask, is it possible for the mask to be applied at runtime?
I've tried playing around with the technologies_masks option in nfcManager_enableDiscovery but it does not seem to do much.

Thanks again!

jgeslin, 2021-06-07:
The Linux libnfc-nci is indeed not designed to allow setting the HCE mode at runtime.

matthew5025, 2021-06-07:
Thanks!

I see that in the example repo there was an issue regarding the setting of UID dynamically.
(https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/issues/2)

Would the patch also work for type b by simply changing LA_NFCID1 (0x33) to the corresponding value for type b?

jgeslin, 2021-06-07:
Yes, this patch can be used to set any NCI parameter (including type B UID) from the application (runtime).

matthew5025, 2021-06-22:
@jgeslin thanks for the help, can confirm it works well.

On the changing of HCE modes, do you think

1. Calling nfcManager_doDeinitialize
2. Editing the config file
3. Calling nfcManager_doInitialize

would allow for dynamic switching of HCE modes?

jgeslin, 2021-06-23:
Yes it will allow switching the HCE modes.

matthew5025, 2021-06-25:
Thanks! 
On a separate note, I've also been trying the tag reader example (https://github.com/NXPNFCLinux/linux_libnfc-nci_examples/blob/master/tag-raw_example/main.c)

What would be the recommend way of retrieving the ATQB response to read information such as application data?

matthew5025, 2021-07-09:
Just a note in the event someone searches for this, I had to modify the library to access the atqb response.

closed 2021-07-09 by matthew5025