NXPNFCLinux/linux_libnfc-nci #133: LPCUSBSIO Hang

matthew5025, 2021-07-12:
In card emulation mode, utilising the lpcusbsio communication option with a response size of larger than ~190 results in the device hanging and no longer responding to any further requests. The USB device has to be removed and reinserted again for it to resume normal operation.

This bug is not seen when utilising the alt communication method.

matthew5025, 2021-07-12:
# Log
```
2021:07:12-01:52:36.433	NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)
2021:07:12-01:52:36.433	NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
2021:07:12-01:52:36.433	NxpHal:     Entering phNxpNciHal_init_monitor
2021:07:12-01:52:36.433	NxpHal:     Returning with SUCCESS
2021:07:12-01:52:36.433	NxpExtns:   find found NXP_NFC_DEV_NODE
2021:07:12-01:52:36.433	NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE
2021:07:12-01:52:36.434	NxpTml:     Total I2C devices: 1
2021:07:12-01:52:36.492	NxpTml:     Handle: 0xac005a60
2021:07:12-01:52:37.185	NxpTml:     PN54X - Tml Reader Thread Started................
2021:07:12-01:52:37.185	NxpTml:     PN54X - Read requested.....
2021:07:12-01:52:37.185	NxpTml:     PN54X - Invoking Read.....
2021:07:12-01:52:37.185	NxpTml:     PN54X - Tml Writer Thread Started................
2021:07:12-01:52:37.185	NxpTml:     PN54X - Tml Writer Thread Running................
2021:07:12-01:52:37.185	NxpTml:     PN54X - Write requested.....
2021:07:12-01:52:37.185	NxpTml:     PN54X - Invoking Write.....
2021:07:12-01:52:37.185	NxpHal:     thread started
2021:07:12-01:52:37.195	NxpNciX:     len =   4 > 20000101
2021:07:12-01:52:37.195	NxpTml:     PN54X - Write successful....
2021:07:12-01:52:37.195	NxpTml:     PN54X - Posting Fresh Write message.....
2021:07:12-01:52:37.195	NxpTml:     PN54X - Tml Writer Thread Running................
2021:07:12-01:52:37.195	NxpHal:     write successful status = 0x0
2021:07:12-01:52:37.195	NxpHal:     Response timer started
2021:07:12-01:52:37.195	NxpHal:     Waiting after ext cmd sent
2021:07:12-01:52:37.221	NxpTml:     I2C_DeviceRead len = 260 returned 6h
2021:07:12-01:52:37.221	NxpTml:     PN54X - Read successful.....
2021:07:12-01:52:37.221	NxpNciR:     len =   6 > 400003001101
2021:07:12-01:52:37.221	NxpTml:     PN54X - Posting read message.....
2021:07:12-01:52:37.221	NxpHal:     read successful status = 0x0
2021:07:12-01:52:37.221	NxpHal:     Response timer stopped
2021:07:12-01:52:37.221	NxpHal:     Checking response
2021:07:12-01:52:37.221	NxpTml:     PN54X - Read requested.....
2021:07:12-01:52:37.221	NxpTml:     PN54X - Invoking Read.....
2021:07:12-01:52:37.221	NxpTml:     PN54X - Write requested.....
2021:07:12-01:52:37.221	NxpTml:     PN54X - Invoking Write.....
2021:07:12-01:52:37.229	NxpNciX:     len =   3 > 200100
2021:07:12-01:52:37.229	NxpTml:     PN54X - Write successful....
2021:07:12-01:52:37.229	NxpTml:     PN54X - Posting Fresh Write message.....
2021:07:12-01:52:37.229	NxpTml:     PN54X - Tml Writer Thread Running................
2021:07:12-01:52:37.229	NxpHal:     write successful status = 0x0
2021:07:12-01:52:37.229	NxpHal:     Response timer started
2021:07:12-01:52:37.229	NxpHal:     Waiting after ext cmd sent
2021:07:12-01:52:37.233	NxpTml:     I2C_DeviceRead len = 260 returned 1ch
2021:07:12-01:52:37.233	NxpTml:     PN54X - Read successful.....
2021:07:12-01:52:37.233	NxpNciR:     len =  28 > 40011900031E030008000102038081828302D002FF020004881001A0
2021:07:12-01:52:37.233	NxpTml:     PN54X - Posting read message.....
2021:07:12-01:52:37.233	NxpHal:     read successful status = 0x0
2021:07:12-01:52:37.233	NxpHal:     NxpNci> FW Version: 10.1.a0
2021:07:12-01:52:37.233	NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
2021:07:12-01:52:37.233	NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
2021:07:12-01:52:37.233	NxpHal:     Response timer stopped
2021:07:12-01:52:37.233	NxpHal:     Checking response
2021:07:12-01:52:37.233	NxpTml:     PN54X - Read requested.....
2021:07:12-01:52:37.233	NxpTml:     PN54X - Invoking Read.....
2021:07:12-01:52:37.233	NxpTml:     PN54X - Write requested.....
2021:07:12-01:52:37.233	NxpTml:     PN54X - Invoking Write.....
2021:07:12-01:52:37.243	NxpNciX:     len =   6 > 20030301A005
2021:07:12-01:52:37.243	NxpTml:     PN54X - Write successful....
2021:07:12-01:52:37.243	NxpTml:     PN54X - Posting Fresh Write message.....
2021:07:12-01:52:37.243	NxpTml:     PN54X - Tml Writer Thread Running................
2021:07:12-01:52:37.243	NxpHal:     write successful status = 0x0
2021:07:12-01:52:37.243	NxpHal:     Response timer started
[... 565 lines cut ...]
2021:07:12-01:52:38.973	NxpTml:     PN54X - Invoking Read.....
2021:07:12-01:52:38.973	NxpFunc:    RoutingManager::stackCallback: event=0x18
2021:07:12-01:52:38.973	NxpFunc:    checkforTranscation: enter; event=0x18 transaction_data.current_transcation_state = 0x0
2021:07:12-01:52:38.973	NxpFunc:    checkforTranscation: exit; event=0x18 transaction_data.current_transcation_state = 0x18
-------------
Card activated - Remote reader is type A
2021:07:12-01:52:38.973	NxpFunc:    RoutingManager::stackCallback: event=0x17
[... cut ...] 3C54EC18DB5C021AFE43FBFAAA3AFB29D1E6053C7C9475D8BE6189F95CBBA8990F95B1EBF1B305EFF700E9A13AE5CA0BCBD0484764BD1F231EA81C7B64C514735AC55E4B79633B706424119E09DCAAD4ACF21B10AF3B33CDE3504847155CBB6F2219BA9B7D
2021:07:12-01:52:39.007	NxpTml:     PN54X - Write successful....
2021:07:12-01:52:39.007	NxpTml:     PN54X - Posting Fresh Write message.....
2021:07:12-01:52:39.007	NxpTml:     PN54X - Tml Writer Thread Running................
2021:07:12-01:52:39.007	NxpHal:     write successful status = 0x0
```

matthew5025, 2021-07-12:
# Code
```c
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <string.h>
#include <pthread.h>
#include "linux_nfc_api.h"

pthread_cond_t condition = PTHREAD_COND_INITIALIZER;
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

nfcHostCardEmulationCallback_t g_HceCB;
unsigned char *HCE_data = NULL;
unsigned int HCE_dataLength = 0x00;
unsigned char g_readerMode = 0x00;

void onDataReceived(unsigned char *data, unsigned int data_length)
{
    printf("Data received\n");

    HCE_dataLength = data_length;
    HCE_data = malloc(HCE_dataLength * sizeof(unsigned char));
    memcpy(HCE_data, data, data_length);

    pthread_cond_signal(&condition);
}

void onHostCardEmulationActivated(unsigned char mode)
{
    printf("-------------\nCard activated - ");
    g_readerMode = mode;
    switch(mode)
    {
        case MODE_LISTEN_A: printf("Remote reader is type A\n");
            break;
        case MODE_LISTEN_B: printf("Remote reader is type B\n");
            break;
        case MODE_LISTEN_F: printf("Remote reader is type F\n");
            break;
        default: printf(" Remote reader type is unknown\n");
            break;
    }
}

void onHostCardEmulationDeactivated()
{
    printf("Card deactivated\n-------------\n");
    g_readerMode = 0x00;
}

int main(int argc, char ** argv) {
    int i;
    int res;
    unsigned char T3tId[]={/* system code*/ 0x01,0xfc, /* nfcid2 */ 0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08};
    unsigned char HCE_response[255];
    unsigned char HCE_response_len;
    srand(0);

    g_HceCB.onDataReceived = onDataReceived;
    g_HceCB.onHostCardEmulationActivated = onHostCardEmulationActivated;
    g_HceCB.onHostCardEmulationDeactivated = onHostCardEmulationDeactivated;

    nfcManager_doInitialize();
    nfcHce_registerHceCallback(&g_HceCB);
    nfcHce_registerT3tIdentifier(T3tId, sizeof(T3tId));
    nfcManager_enableDiscovery(0x00, 0, 1, 0);

    HCE_response_len = 190;

    /* Just reply with status OK */
    for (i = 0; i < HCE_response_len; i++) {
        HCE_response[i] = rand();
    }

    printf("\n-------------\nWaiting for reader...\n");
    do{
        /* Wait for data from remote reader */
        pthread_cond_wait(&condition, &mutex);

    if(HCE_data != NULL)
    {
        printf("\t\tReceived data from remote device : \n\t\t");
        for(i = 0x00; i < HCE_dataLength; i++)
        {
            printf("%02X ", HCE_data[i]);
        }

        switch (g_readerMode)
        {
            case MODE_LISTEN_A:
            case MODE_LISTEN_B:
            case MODE_LISTEN_F:
                break;

            default:
                res = 0x00;
                break;
            }

            if(HCE_response_len > 0)
            {
                res = nfcHce_sendCommand(HCE_response, HCE_response_len);
            }
            else
            {
                res = 0;
            }

            if(0x00 == res)
            {
                printf("\n\n\t\tResponse sent : \n\t\t");
                for(i = 0x00; i < HCE_response_len; i++)
                {
                    printf("%02X ", HCE_response[i]);
                }
                printf("\n\n");
                printf("Response length is %d\n", HCE_response_len);
            }
            else
            {
                printf("\n\n\t\tFailed to send response\n\n");
            }
        }
    }while(1);
}
```

jgeslin, 2021-08-11:
Hi,
Sorry for the late response.

The lpcusbsio implementation is using PN7150 I2C fragmentation mechanism and thus need to be enabled within libnfc-nxp-init.conf file.

Please try out enabling it (NXP_I2C_FRAGMENTATION_ENABLED=0x01) and feedback.

Regards,
Jeremy.

matthew5025, 2021-08-20:
@jgeslin thanks for the reply! 
Unfortunately setting NXP_I2C_FRAGMENTATION_ENABLED=0x01 still results in the same issue for me

jgeslin, 2021-09-02:
Can you please provide logs (initialization step) after updating the configuration file with NXP_I2C_FRAGMENTATION_ENABLED=0x01 ?

matthew5025, 2021-09-02:
```
tail -1 /usr/local/etc/libnfc-nxp-init.conf  
NXP_I2C_FRAGMENTATION_ENABLED=0x01
```
**Log**
```
2021:09:03-00:56:24.002	NxpExtns:   find found NXPLOG_TML_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_TML_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   find found NXPLOG_FWDNLD_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_FWDNLD_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   find found NXPLOG_NCIX_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIX_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   find found NXPLOG_NCIR_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpExtns:   GetNxpNumValue: NXP Config Parameter : NXPLOG_NCIR_LOGLEVEL=(0x3)
2021:09:03-00:56:24.002	NxpFunc:    phNxpLog_InitializeLogLevel: global =3, Fwdnld =3, extns =3,                     hal =3, tml =3, ncir =3,                     ncix =3
2021:09:03-00:56:24.002	NxpHal:     Entering phNxpNciHal_init_monitor
2021:09:03-00:56:24.002	NxpHal:     Returning with SUCCESS
2021:09:03-00:56:24.002	NxpExtns:   find found NXP_NFC_DEV_NODE
2021:09:03-00:56:24.002	NxpExtns:   GetNxpStrValue: NXP Config Parameter : NXP_NFC_DEV_NODE
2021:09:03-00:56:24.003	NxpTml:     Total I2C devices: 1
2021:09:03-00:56:24.059	NxpTml:     Handle: 0xd0005480
2021:09:03-00:56:24.420	NxpTml:     PN54X - Tml Reader Thread Started................
2021:09:03-00:56:24.420	NxpTml:     PN54X - Read requested.....
2021:09:03-00:56:24.420	NxpTml:     PN54X - Invoking Read.....
2021:09:03-00:56:24.420	NxpTml:     PN54X - Tml Writer Thread Started................
2021:09:03-00:56:24.420	NxpTml:     PN54X - Tml Writer Thread Running................
2021:09:03-00:56:24.420	NxpTml:     PN54X - Write requested.....
2021:09:03-00:56:24.420	NxpTml:     PN54X - Invoking Write.....
2021:09:03-00:56:24.420	NxpHal:     thread started
2021:09:03-00:56:24.426	NxpTml:     I2C_DeviceRead len = 260 returned 4h
2021:09:03-00:56:24.426	NxpTml:     PN54X - Read successful.....
2021:09:03-00:56:24.426	NxpNciR:     len =   4 > 600701A1
2021:09:03-00:56:24.426	NxpTml:     PN54X - Posting read message.....
2021:09:03-00:56:24.426	NxpHal:     read successful status = 0x0
2021:09:03-00:56:24.426	NxpTml:     PN54X - Read requested.....
2021:09:03-00:56:24.426	NxpTml:     PN54X - Invoking Read.....
2021:09:03-00:56:24.427	NxpNciX:     len =   4 > 600701A1
2021:09:03-00:56:24.427	NxpTml:     PN54X - Write successful....
2021:09:03-00:56:24.428	NxpTml:     PN54X - Posting Fresh Write message.....
2021:09:03-00:56:24.428	NxpTml:     PN54X - Tml Writer Thread Running................
2021:09:03-00:56:24.428	NxpHal:     write successful status = 0x0
2021:09:03-00:56:24.428	NxpHal:     Response timer started
2021:09:03-00:56:24.428	NxpHal:     Waiting after ext cmd sent
2021:09:03-00:56:24.428	NxpHal:     Response timer stopped
2021:09:03-00:56:24.428	NxpHal:     Checking response
2021:09:03-00:56:24.428	NxpTml:     PN54X - Write requested.....
2021:09:03-00:56:24.428	NxpTml:     PN54X - Invoking Write.....
2021:09:03-00:56:24.429	NxpTml:     I2C_DeviceRead len = 260 returned 4h
2021:09:03-00:56:24.429	NxpTml:     PN54X - Read successful.....
2021:09:03-00:56:24.429	NxpNciR:     len =   4 > 600701A1
2021:09:03-00:56:24.429	NxpTml:     PN54X - Posting read message.....
2021:09:03-00:56:24.430	NxpHal:     read successful status = 0x0
2021:09:03-00:56:24.430	NxpTml:     PN54X - Read requested.....
2021:09:03-00:56:24.430	NxpTml:     PN54X - Invoking Read.....
2021:09:03-00:56:24.439	NxpTml:     _i2c_write() errno : 0
2021:09:03-00:56:24.440	NxpTml:     PN54X - Error in Write.....
2021:09:03-00:56:24.440	NxpTml:     PN54X - Posting Fresh Write message.....
2021:09:03-00:56:24.440	NxpTml:     PN54X - Tml Writer Thread Running................
2021:09:03-00:56:24.440	NxpHal:     write error status = 0x1ff
2021:09:03-00:56:24.440	NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
2021:09:03-00:56:24.442	NxpTml:     PN54X - Write requested.....
2021:09:03-00:56:24.442	NxpTml:     PN54X - Invoking Write.....
2021:09:03-00:56:24.454	NxpNciX:     len =   3 > 200100
2021:09:03-00:56:24.454	NxpTml:     PN54X - Write successful....
2021:09:03-00:56:24.454	NxpTml:     PN54X - Posting Fresh Write message.....
2021:09:03-00:56:24.454	NxpTml:     PN54X - Tml Writer Thread Running................
2021:09:03-00:56:24.454	NxpHal:     write successful status = 0x0
2021:09:03-00:56:24.454	NxpHal:     Response timer started
2021:09:03-00:56:24.454	NxpHal:     Waiting after ext cmd sent
2021:09:03-00:56:24.454	NxpHal:     Response timer stopped
[... 601 lines cut ...]
2021:09:03-00:56:24.986	NxpHal:     > Polling Loop Started
2021:09:03-00:56:24.986	NxpTml:     PN54X - Write requested.....
2021:09:03-00:56:24.986	NxpTml:     PN54X - Invoking Write.....
2021:09:03-00:56:24.993	NxpNciX:     len =  10 > 21030703800181018201
2021:09:03-00:56:24.993	NxpTml:     PN54X - Write successful....
2021:09:03-00:56:24.993	NxpTml:     PN54X - Posting Fresh Write message.....
2021:09:03-00:56:24.993	NxpTml:     PN54X - Tml Writer Thread Running................
2021:09:03-00:56:24.993	NxpHal:     write successful status = 0x0
2021:09:03-00:56:24.995	NxpTml:     I2C_DeviceRead len = 260 returned 4h
2021:09:03-00:56:24.995	NxpTml:     PN54X - Read successful.....
2021:09:03-00:56:24.995	NxpNciR:     len =   4 > 41030100
[... cut ...]
2021:09:03-00:56:24.996	NxpFunc:    nfaConnectionCallback: event= 30
2021:09:03-00:56:24.996	NxpFunc:    nfaConnectionCallback: NFA_RF_DISCOVERY_STARTED_EVT: status = 0
2021:09:03-00:56:24.996	NxpFunc:    nativeNfcTag_releaseRfInterfaceMutexLock: sRfInterfaceMutex unlock
2021:09:03-00:56:24.996	NxpFunc:    nativeNfcManager_enableDiscovery: exit

-------------
Waiting for reader...
2021:09:03-00:56:24.996	NxpTml:     PN54X - Read requested.....
2021:09:03-00:56:24.996	NxpTml:     PN54X - Invoking Read.....
^C
```

cvt1982, 2022-08-16:
Hi,
I have the same issue. Were you ever able to find a solution to the problem?

matthew5025, 2022-08-24:
@cvt1982 Nope sorry, I never did

cvt1982, 2022-08-24:
@matthew5025 
I have made a support ticket at NXP where I have linked to this issue. I will return with the answer here if I ever get one.