NXPNFCLinux/linux_libnfc-nci #139: First Issue Posted: Raw Read works, but initialization has many errors, and phFriNfc_MifareStdMap_ChkNdef fails

kwmartin, 2022-01-14:
I've never posted an issue; will try to follow best practices. Not sure how to include a link to code which would make this issue much smaller.
Raspberry Pi 4b, Custom board with PN1750 (not sure which version - it was hard to get them with chip shortage).
`[New Thread 0xfffff61a4900 (LWP 9793)]
2022:01:14-13:54:43.751	NxpTml:     System already has pin /sys/class/gpio/gpio12
2022:01:14-13:54:43.751	NxpTml:     System already has pin /sys/class/gpio/gpio16
[New Thread 0xfffff71a71c0 (LWP 9794)]
[New Thread 0xfffff61821c0 (LWP 9795)]
[New Thread 0xfffff59811c0 (LWP 9796)]
2022:01:14-13:54:43.982	NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
2022:01:14-13:54:43.983	NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
2022:01:14-13:54:43.985	NxpHal:     i2c_fragmentation_status existing
2022:01:14-13:54:43.985	NxpFwDnld:  Ignoring Config :NFC FW Binary Path:
2022:01:14-13:54:43.985	NxpFwDnld:  Ignoring Config:NFC FW Binary :libpn548ad
2022:01:14-13:54:43.985	NxpFwDnld:  NFC FW Binary :/libpn548ad_fw.so
2022:01:14-13:54:43.985	NxpFwDnld:  NULL handler : unable to load the binary file, verify binary path
2022:01:14-13:54:43.985	NxpFwDnld:  Image extraction Failed - invalid imginfo or imginfolen!!
2022:01:14-13:54:43.986	NxpFwDnld:  Error loading libpn54x_fw !!
2022:01:14-13:54:43.986	NxpFwDnld:  Freeing Mem for Dnld Context..
2022:01:14-13:54:43.993	NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
2022:01:14-13:54:43.993	NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
2022:01:14-13:54:43.995	NxpFwDnld:  Clock frequency value is wrong in config file, setting it as default
2022:01:14-13:54:44.059	NfcHcpX:8103
2022:01:14-13:54:44.062	NfcHcpR:8180
2022:01:14-13:54:44.062	NfcHcpX:810103020304
2022:01:14-13:54:44.065	NfcHcpR:8180
2022:01:14-13:54:44.065	NfcHcpX:810101f51f7e66f21f7e66
2022:01:14-13:54:44.074	NfcHcpR:8180
2022:01:14-13:54:44.221	NfcHcpX:810204
2022:01:14-13:54:44.224	NfcHcpR:818000
2022:01:14-13:54:44.225	NxpFunc:    initialize:HOST_LISTEN_TECH_MASK=7;

`
The i2c after these initialization errors seems to be working correctly. My tags are classic 1K, I verified them with NXP Info App.
The tags have been formatted to be NDEF, but the NDEF checking mostly fails. I'm currently running debug level 2 as level3 gives so much info, I can't figure out what is going on. I have debugging in gdb working so I can break most anywhere.

When things go wrong I get a lot of
`Tag not yet NDEF formatted, trying to format
2022:01:14-13:32:55.260	NxpExtns:   Error Sending msg to Extension Thread
2022:01:14-13:32:55.274	NxpExtns:   Error Sending msg to Extension Thread
`
Even though tag is ndef formatted.  When running raw read, I get
`> tag-raw_example
2022:01:14-14:23:31.564	NxpTml:     System already has pin /sys/class/gpio/gpio12
2022:01:14-14:23:31.564	NxpTml:     System already has pin /sys/class/gpio/gpio16
2022:01:14-14:23:31.766	NxpTml:     _i2c_write() errno : 79
2022:01:14-14:23:31.766	NxpTml:     PN54X - Error in I2C Write.....
2022:01:14-14:23:31.766	NxpHal:     write error status = 0x1ff
2022:01:14-14:23:31.767	NxpHal:     write_unlocked failed - PN54X Maybe in Standby Mode - Retry
2022:01:14-14:23:31.796	NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
2022:01:14-14:23:31.799	NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
2022:01:14-14:23:31.801	NxpHal:     i2c_fragmentation_status existing
2022:01:14-14:23:31.801	NxpFwDnld:  Ignoring Config :NFC FW Binary Path:
2022:01:14-14:23:31.802	NxpFwDnld:  Ignoring Config:NFC FW Binary :libpn548ad
2022:01:14-14:23:31.802	NxpFwDnld:  NFC FW Binary :/libpn548ad_fw.so
2022:01:14-14:23:31.802	NxpFwDnld:  NULL handler : unable to load the binary file, verify binary path
2022:01:14-14:23:31.802	NxpFwDnld:  Image extraction Failed - invalid imginfo or imginfolen!!
2022:01:14-14:23:31.802	NxpFwDnld:  Error loading libpn54x_fw !!
2022:01:14-14:23:31.802	NxpFwDnld:  Freeing Mem for Dnld Context..
2022:01:14-14:23:31.810	NxpHal:     phNxpNciHal_deriveChipType HwVersion : 0x88
2022:01:14-14:23:31.811	NxpHal:     phNxpNciHal_deriveChipType NxpNci > Product : PN548C2
2022:01:14-14:23:31.813	NxpFwDnld:  Clock frequency value is wrong in config file, setting it as default
2022:01:14-14:23:31.879	NfcHcpX:8103
2022:01:14-14:23:31.882	NfcHcpR:8180
2022:01:14-14:23:31.883	NfcHcpX:810103020304
2022:01:14-14:23:31.886	NfcHcpR:8180
2022:01:14-14:23:31.886	NfcHcpX:810101e4c28066e1c28066
2022:01:14-14:23:31.895	NfcHcpR:8180
2022:01:14-14:23:32.045	NfcHcpX:810204
2022:01:14-14:23:32.049	NfcHcpR:818000
2022:01:14-14:23:32.051	NxpFunc:    initialize:HOST_LISTEN_TECH_MASK=7;

-------------
Waiting for tag...
2022:01:14-14:23:36.445	NxpFunc:    NfcTag::processNotification (Activation): RF disc_id=01; protocol=80, mNumDiscList=0
2022:01:14-14:23:36.445	NxpFunc:    Mifare Classic detected
2022:01:14-14:23:36.445	NxpFunc:    processNotification (activation): NumTechList = 1 sel_rsp=0x8 
2022:01:14-14:23:36.445	NxpFunc:    NfcTag::processNotification; mNumTechList=2
2022:01:14-14:23:36.445	NxpFunc:    NfcTag::discoverTechnologies (activation): index=0; tech=1; handle=1; nfc type=80
2022:01:14-14:23:36.446	NxpFunc:    NfcTag::discoverTechnologies (activation): index=1; tech=8; handle=1; nfc type=80
Tag detected
MifareClassic Authenticate command: 60 
01 
00 
00 
00 
00 
FF 
FF 
FF 
FF 
FF 
FF 
MifareClassic Authenticate command sent - Response: 40 

03 

MifareUL Read command: 30 
01 
MifareUL Read command sent - Response: 10 

04 

00 

Tag removed
-------------

-------------
Waiting for tag...
2022:01:14-14:23:37.658	NxpFunc:    connectionEventHandler:NFC Tag/Target Deactivated

`
which looks correct to me except for the initialization errors. I have no idea yet what is causing the initialization errors. I checked the enbable GPIO using gpio readall and it states output and '1'. I'll leave it at this for now, but if anyone has any suggestions, I am very open to trying them. Thanks.

kwmartin, 2022-01-14:
P.S. Can anyone direct me to a link for the command for a raw write. I'm guessing I could use raw reads and writes and not use ndef, although when ndef works, it's much simpler.

jgeslin, 2022-01-17:
Hi, 

The error logs you are getting during initialization are expected. Indeed it relates to a NFC Controller FW update procedure, not supported by PN7150, implemented for other NFC Controllers (PN548 for instance).
The issue you are facing about "Tag not NDEF" while tag is NDEF formatted is definitely not linked to this failing procedure.

To further analyze this issue, you need to look to more detailled debug traces. This is enabled by settings all NXPLOG_xxx_LOGLEVEL flags to 0x03 in libnfc-nxp-init.conf file.
It shall allow to understand why the libnfc-nci library is reporting the tag as not NDEF.

Regards,
Jeremy.

kwmartin, 2022-01-17:
I will do. Is the best way to include the log outputs to drag a file here. I am also trying to do raw reads and writes as in the examples. The raw read returns 16 bytes read, but there values are not correct. Is it necessary to "unformat" an ndef tag before one can do raw reads and writes? Is it possible to look at all the data sent and received including the CRC's. It would be very nice to have a complete raw write and read example as a starting point.

Here is relevant part of where I changed the raw write example to include a write to block one before a read back.
```
            case TARGET_TYPE_MIFARE_CLASSIC:
            {
                unsigned char AuthCmd[] = {0x60U, /*block*/ 0x01, 0x00, 0x00, 0x00, 0x00, /*key*/ 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
                unsigned char ReadCmd[] = {0x30U, /*block*/ 0x00};
				unsigned char WriteCmd[] = {0xA0, 0x01, 0x0,0x1,0x2,0x3,0x4,0x5,0x6,0x7,0x8,0x9,0xA,0xB,0xC,0xD,0xE,0xF};
                unsigned char Resp[20];

                printf("MifareClassic Authenticate command: ");
                for(i = 0x00; i < (unsigned int) sizeof(AuthCmd) ; i++) { printf("%02X ", AuthCmd[i]); printf("\n"); }
                res = nfcTag_transceive(g_tagInfos.handle, AuthCmd, 12, Resp, sizeof(Resp), 500);
                if(0x00 == res)
                {
                    printf("\n\t\tRAW Tag transceive failed\n");
                }
                else {
                    printf("MifareClassic Authenticate command sent - Response: ");
                    for(i = 0x00; i < (unsigned int)res; i++) { printf("%02X ", Resp[i]); printf("\n\n"); }
                }

				usleep(50000);
                printf("MifareUL Write command: ");
                for(i = 0x00; i < (unsigned int) sizeof(WriteCmd) ; i++) { printf("\n\tWriteCmd[%d]: 0x%02X\n", i, WriteCmd[i]); }
                res = nfcTag_transceive(g_tagInfos.handle, WriteCmd, sizeof(WriteCmd), Resp, sizeof(Resp), 500);
                if(0x00 == res) {
                    printf("RAW Tag transceive failed\n");
                }
                else {
                    printf("MifareUL Write command sent - Response: ");
                    for(i = 0x00; i < (unsigned int)res; i++) { printf("\n\tResp[%d]: 0x%02X\n", i, Resp[i]); }
                }

				usleep(50000);
                printf("MifareUL Read command: ");
                for(i = 0x00; i < (unsigned int) sizeof(ReadCmd) ; i++) { printf("0x%02X ", ReadCmd[i]); printf("\n"); }
                res = nfcTag_transceive(g_tagInfos.handle, ReadCmd, sizeof(ReadCmd), Resp, sizeof(Resp), 500);
                if(0x00 == res) {
                    printf("RAW Tag transceive failed\n");
                }
                else {
                    printf("MifareUL Read command sent - Response: ");
                    for(i = 0x00; i < (unsigned int)res; i++) { printf("\n\tResp[%d]: 0x%02X\n", i, Resp[i]);}
                }
                fflush(stdout);
				a = 1;
            }

            break;
```
[raw.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/7884798/raw.log)
From the log, it appears the Read command is only returning 2 bytes {0x10, 0xB2} rather than a  16byte block. Yet the logs seem to indicate the raw read was successful

```
MifareUL Read command: 0x30 
0x00 
2022:01:17-16:38:25.314	NxpFunc:    nativeNfcTag_doTransceive: enter
2022:01:17-16:38:25.314	NxpFunc:    Nfc: NFA_SendRawFrame () data_len:3
2022:01:17-16:38:25.314	NxpFunc:    Nfc: NFA got event 0x010C
2022:01:17-16:38:25.314	NxpFunc:    Nfc: nfa_dm_evt_hdlr event: 0xc
2022:01:17-16:38:25.314	NxpFunc:    Nfc: nfa_dm_act_send_raw_frame ()
2022:01:17-16:38:25.314	NxpFunc:    Nfc: nfc_ncif_send_data :0, num_buff:1 qc:0
2022:01:17-16:38:25.315	NxpFunc:    NfcAdaptation::HalWrite
2022:01:17-16:38:25.315	NxpTml:     PN54X - Write requested.....
2022:01:17-16:38:25.315	NxpTml:     PN54X - Invoking I2C Write.....
2022:01:17-16:38:25.316	NxpNciX:     len =   6 > 000003103000
2022:01:17-16:38:25.316	NxpTml:     PN54X - I2C Write successful.....
2022:01:17-16:38:25.316	NxpTml:     PN54X - Posting Fresh Write message.....
[... 19 lines cut ...]
2022:01:17-16:38:25.328	NxpTml:     PN54X - Read requested.....
2022:01:17-16:38:25.328	NxpTml:     PN54X - Invoking I2C Read.....
2022:01:17-16:38:25.328	NxpFunc:    Nfc: Process Event mt:0
2022:01:17-16:38:25.328	NxpFunc:    Nfc: NFC received data
2022:01:17-16:38:25.328	NxpFunc:    Nfc: nfc_ncif_proc_data 0x000002
2022:01:17-16:38:25.328	NxpFunc:    Nfc: nfc_ncif_proc_data len:2
2022:01:17-16:38:25.328	NxpFunc:    Nfc: nfa_rw_raw_mode_data_cback(): event = 0x6003
2022:01:17-16:38:25.328	NxpFunc:    nfaConnectionCallback: event= 9
2022:01:17-16:38:25.328	NxpFunc:    nfaConnectionCallback: NFA_DATA_EVT: status = 0x0, len = 2
2022:01:17-16:38:25.328	NxpFunc:    nativeNfcTag_doTransceiveStatus: data len=2
2022:01:17-16:38:25.328	NxpFunc:    nativeNfcTag_doTransceive: response 2 bytes
2022:01:17-16:38:25.328	NxpFunc:    nativeNfcTag_doTransceive: exit
MifareUL Read command sent - Response: 
	Resp[0]: 0x10
2022:01:17-16:38:25.329	
NxpFunc:    	Resp[1]: 0xB2
presenceCheckThread: Presence Check - Scheduled

Thread 1 "tag-raw_example" hit Breakpoint 1, main (argc=1, argv=0xffffffffecf8) at main.c:179
179					a = 1;
(gdb) x/2xb Resp
0xffffffffea98:	0x10	0xb2
(gdb) 

```

jgeslin, 2022-01-18:
Raw command is definitely not the best choice when working with NDEF. Indeed, depending of the tag type you must handle all commands to respect NFC Forum specification. This is where the libnfc-nci library helps by abstracting all this stuff.
Please forward log of the failing scenario using NDEF interface, so I can check what is going wrong using this interface.

kwmartin, 2022-01-18:
Will do, but currently I an trying to get raw reads and writes working where I can have control on what is happening. The NDEF tags take a lot more time and space which my application doesn't need. Because they are much larger, I find they are much more prone to having errors. If you could help let me know why the simple raw write and read is failing, I would really appreciate it. I understand that ibnfc-nci library helps by abstracting all this stuff, but when it fails, this abstraction makes debugging almost impossible; if it works, all is good; if it fails, one is mostly out of luck. When I previously used PN532, I was able to get raw writes and reads working well; I am not having the the same success with the PN1750. I also seem to have a number of PN1750s with broken I2C interfaces even though I added 50 series resistors to protect the inputs. Are there versions of the PN1750 that have problems and should be avoided>