NXPNFCLinux/linux_libnfc-nci #141: pn7150 returns  the same id tag if I have passed another badge after passed MIFARE CLASSIC card

jorisoffouga, 2022-03-21:
When passing a badge on the RFID field with a MIFARE CLASSIC type card, the pn7150 returns  the same id tag even if I have passed another badge can you give me some leads for the resolution of this bug?

CITIZENGATE, 2022-03-24:
Hi,
I have exactly the same trouble. On my side it not appends immediatly but after 5 or 6 badges.
I'm using the kernel driver.

skupekA, 2022-03-30:
Hi,
can you please enable logs in file "usr/local/etc/libnfc-nxp.conf" like in image and post the logs.
[image]

Best regards!
Andraz

jorisoffouga, 2022-03-31:
Hi @CITIZENGATE Is same of my side.

Hi @skupekA please you can find log in attached file: [nfcDemo_log.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/8390869/nfcDemo_log.txt)

jgeslin, 2022-04-04:
Hi @jorisoffouga, @CITIZENGATE,

Could you please provide complete logs of the scenario (presenting successively different cards showing the same UID), including the application output (Tag UID value displayed by the application) and the libnfc-nci traces as shown in below example ?

```
2022:04:04-13:42:38.586 NxpFunc:    NfcTag::discoverTechnologies (activation): index=1; tech=8; handle=1; nfc type=80
2022:04:04-13:42:38.586 NxpFunc:    NfcTag::discoverTechnologies (activation): exit
2022:04:04-13:42:38.587 NxpFunc:    NfcTag::createNativeNfcTag: enter
2022:04:04-13:42:38.587 NxpFunc:    createNativeNfcTag: Selected: index=1; tech=8; handle=1; nfc type=80
2022:04:04-13:42:38.587 NxpFunc:    NfcTag::setNfcTagUid: tech A
2022:04:04-13:42:38.588 NxpFunc:    nativeNfcTag_onTagArrival: notify tag is ready
        NFC Tag Found

        Type :         'Type A - Mifare Classic'
        NFCID1 :        'F6 19 05 01 '
2022:04:04-13:42:38.588 NxpFunc:    nativeNfcTag_checkNdef: enter; handle=1
2022:04:04-13:42:38.588 NxpFunc:    nativeNfcTag_doReconnect: enter
2022:04:04-13:42:38.588 NxpFunc:    nativeNfcTag_doReconnect: enter; handle=1
2022:04:04-13:42:38.589 NxpFunc:    reSelect: enter; rf intf = 128, current intf = 128
2022:04:04-13:42:38.589 NxpFunc:    setReconnectState = 0x0
2022:04:04-13:42:38.589 NxpFunc:    reSelect: deactivate to sleep
```

Regards,
Jeremy.