NXPNFCLinux/linux_libnfc-nci #143: RF_DISCOVER_CMD returns STATUS_SYNTAX_ERROR response

mwalle, 2022-07-06:
Hi,

I'm trying to get a PN7160 to work with an IMX8MN. I've downloaded this repo, checked out the NCI2.0_R1.1 tag, build the nxpnfc kernel module. Basic communication works and the firmware was also successfully update.
The linux_NfcFactoryTestApp works and recognizes tags.

But the demo app of the libnfc-nci library doesn't work. I've traced it and it seems to hang while waiting for a response of the RF_DISCOVER_CMD. That is sNfaEnableDisablePollingEvent.wait() in startRfDiscovery() is blocking.

I've traced the I2C communication and it looks like the RF_DISCOVERY_CMD returns an STATUS_SYNTAX_ERROR response. I'd expect a STATUS_OK response.

```
2022:07:06-14:50:24.258  len =  22 > 21031309000101010201030180018101060174017701
2022:07:06-14:50:24.259  len =   4 > 41030105
```

The command looks ok. Any idea what is going wrong here?

Full log with i2c tracing enabled:
```
# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

2022:07:06-14:50:24.086  len =   4 > 20000100
2022:07:06-14:50:24.091  len =   4 > 40000100
2022:07:06-14:50:24.094  len =  12 > 600009020020040471125006
2022:07:06-14:50:24.098  len =   5 > 2001020000
2022:07:06-14:50:24.100  len =  33 > 40011E001A7E060301D002FFFF01FF000800000100020003008000820083008400
2022:07:06-14:50:24.101  len =   6 > 20030301A005
2022:07:06-14:50:24.102  len =   9 > 4003060001A0050100
2022:07:06-14:50:24.107  len =   5 > 2003020100
2022:07:06-14:50:24.108  len =   9 > 40030600010002F401
2022:07:06-14:50:24.110  len =   4 > 20000100
2022:07:06-14:50:24.114  len =   4 > 40000100
2022:07:06-14:50:24.115  len =  12 > 600009020020040471125006
2022:07:06-14:50:24.117  len =   5 > 2001020000
2022:07:06-14:50:24.118  len =  33 > 40011E001A7E060301D002FFFF01FF000800000100020003008000820083008400
2022:07:06-14:50:24.119  len =   3 > 2F0200
2022:07:06-14:50:24.121  len =   8 > 4F0205000001C113
2022:07:06-14:50:24.122  len =   8 > 20020501A0070103
2022:07:06-14:50:24.123  len =   5 > 4002020000
2022:07:06-14:50:24.126  len =   6 > 20030301A00F
2022:07:06-14:50:24.129  len =  40 > 4003250001A00F200000000000000000000000000000000000000000000000000000000000000000
2022:07:06-14:50:24.130  len =  10 > 20030703A002A003A004
2022:07:06-14:50:24.133  len =  17 > 40030E0003A0020101A0030108A0040101
2022:07:06-14:50:24.135  len =   8 > 20020501A0030108
2022:07:06-14:50:24.137  len =   5 > 4002020000
2022:07:06-14:50:24.138  len =   8 > 20020501A0440100
2022:07:06-14:50:24.139  len =   5 > 4002020000
2022:07:06-14:50:24.140  len =  18 > 20020F01A00E0B110101010000004000D00C
2022:07:06-14:50:24.141  len =   5 > 4002020000
2022:07:06-14:50:24.144  len =  79 > 20024C09A00D03780D02A00D03781402A00D064C4465090000A00D064C2D05351E01A00D06824A55070007A00D0644440304C400A00D06463050001800A00D06483050001800A00D064A3050000800
2022:07:06-14:50:24.150  len =   5 > 4002020000
2022:07:06-14:50:24.152  len =  33 > 20021E03A0380424100B00A0AA0438049808A0AF0C83D578800083D57880007708
2022:07:06-14:50:24.154  len =   5 > 4002020000
2022:07:06-14:50:24.161  len =  94 > 20025B01A00B57F19190500F4E004B15B7AA4B1FA799619F97996C9F97996F9F97997A9F97007C9F0700871F07008C9F0700949F0700979F0700A29F0400A49F0400AF9F0400B19F0400BC9F0200C59F0000CC9F0000D71F0000E21F0000
2022:07:06-14:50:24.162  len =   5 > 4002020000
2022:07:06-14:50:24.168  len = 155 > 20029801A03494230418CB0000590100D90100440200A5020078030045040020050077060077060077060077060077060077060077060077060077060077060077060077060077060077060077060077060018CB0000590100D90100440200A50200780300450400200500770600770600770600770600770600770600770600770600770600770600770600770600770600770600770600770600
2022:07:06-14:50:24.169  len =   5 > 4002020000
2022:07:06-14:50:24.176  len = 167 > 2002A401A0A9A000C1000A0180410A0281830A03C042060480460605C3010306C2050307C24A030841000A08C0D40608C38B030901820A09C392030A4002060AC4CC030B0006060B40D4030C4400030D4204030E4248030F435003104391031144950312468E011347C4011448CA011549D001164BD401174DD60118497203194E68011A5270011B5374011C54B4011D52EA011E53F0011F53F0001F53F0001F53F0001F53F000
2022:07:06-14:50:24.177  len =   5 > 4002020000
2022:07:06-14:50:24.178  len =  12 > 20020902A05E0101A0400100
2022:07:06-14:50:24.179  len =   5 > 4002020000
2022:07:06-14:50:24.180  len =   6 > 20030301A085
2022:07:06-14:50:24.182  len =  12 > 4003090001A085045008082C
2022:07:06-14:50:24.183  len =   8 > 20020501A0950101
2022:07:06-14:50:24.185  len =   5 > 4002020000
2022:07:06-14:50:24.187  len =  39 > 20022401A00F200000000000000000000000000000000000000000000000000000000000000000
2022:07:06-14:50:24.188  len =   5 > 4002020000
2022:07:06-14:50:24.191  len =  52 > 2002310F8501012801002101003001083101033201603801013304010203045401065001025B010080010181010182010E180101
2022:07:06-14:50:24.208  len =   5 > 4002020000
2022:07:06-14:50:24.211  len =  13 > 21000A03040302050303800180
2022:07:06-14:50:24.218  len =   4 > 41000100
2022:07:06-14:50:24.219  len =   3 > 220000
2022:07:06-14:50:24.219  len =   5 > 4200020001
2022:07:06-14:50:24.220  len =  17 > 62000E10010100010406880000003B0101
2022:07:06-14:50:24.221  len =   5 > 2201021001
2022:07:06-14:50:24.222  len =   4 > 42010100
2022:07:06-14:50:24.223  len =   4 > 62010100
2022:07:06-14:50:24.224  len =  14 > 610A0B0200031080040003108104
2022:07:06-14:50:24.226  len =   5 > 2003020152
2022:07:06-14:50:24.227  len =   8 > 4003050001520104
2022:07:06-14:50:24.229  len =   7 > 20020401850101
2022:07:06-14:50:24.230  len =   5 > 4002020000
2022:07:06-14:50:24.232  len =  31 > 21011C000402091039D2760000850101010300010301030011040103000105
2022:07:06-14:50:24.233  len =   4 > 41010100
2022:07:06-14:50:24.235  len =  26 > 20021701291446666D010112020207FF03020003040164070103
2022:07:06-14:50:24.236  len =   5 > 4002020000
2022:07:06-14:50:24.238  len =  19 > 20021005300104310100320120380101500100
2022:07:06-14:50:24.251  len =   5 > 4002020000
2022:07:06-14:50:24.255  len =   8 > 200205010002F401
2022:07:06-14:50:24.256  len =   5 > 4002020000
2022:07:06-14:50:24.258  len =  22 > 21031309000101010201030180018101060174017701
2022:07:06-14:50:24.259  len =   4 > 41030105
2022:07:06-14:50:24.261  len =   4 > 21060100
2022:07:06-14:50:24.261  len =   4 > 41060106
```

skupekA, 2022-07-26:
Hi,

can you please change 21031309000101010201030180018101060174017701 command to 21031309000101010201030180018101060174017**0**01 and try again? Also we will check again our patch, since this issue was already resolved.

Regards,
Andraz

skupekA, 2022-08-22:
Hi,

you can change it in libnfc-nci.conf

Regards,
Andraz

mwalle, 2022-08-22:
Hi,

can you please tell me what I should change there?

Thank,
-michael

skupekA, 2022-08-22:
Hi,

yes, NFA_PROPRIETARY_CFG={05:FF:FF:06:81:80:**70**:FF:FF}

Regards,
Andraz

mwalle, 2022-08-23:
Hi,

that setting already has the correct setting in my configuration file:

NFA_PROPRIETARY_CFG={05, FF, FF, 06, 81, 80, 70, FF, FF}

The nfcDemoApp still sends the 77.

Please note, that this will change if I apply the patch from #144.

-michael

skupekA, 2022-08-23:
Hi,

we are preparing now updated version of Application note for Linux on PN7160. We tried patch
[linux_64bitOs.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9400933/linux_64bitOs.zip) on 64bit version of Linux (Raspberry pi) and everything is working on our site.
1) Please use  latest release with (ee8ad71123b07b5a33012df0c614ab3e9278c080) commit (Add KOVIO support).
2) Apply patch shared in this message
3) There should not be any issue anymore

Kovio support was added couple of months ago. All issues were solved back then.

Regards

mwalle, 2022-08-23:
Hi,

did you test with a non-blank NFC tag as mentioned in #144 ?

This still doesn't work:
```
# nfcDemoApp poll
#########################################################################################
##                                       NFC demo                                      ##
#########################################################################################
##                                 Poll mode activated                                 ##
#########################################################################################
                              ... press enter to quit ...

Waiting for a Tag/Device...   

        NFC Tag Found

        Type :         'Type A - Mifare Ul'
        NFCID1 :        '04 EF C6 AA 28 63 80 '
                Record Found :
                                NDEF Content Max size :     '137 bytes'
                                NDEF Actual Content size :     '11 bytes'
                                ReadOnly :                      'FALSE'
Segmentation fault (core dumped)
```

Also, will you import that patch into the repository?

-michael

skupekA, 2022-08-23:
Hi,

yes it was tested (attached image):
[image]

Yes, as soon as we will prepare AN we will also publish patch to GitHub

Regards,
Andraz

skupekA, 2022-08-23:
What you can try:
APPL_TRACE_LEVEL=0xFF
PROTOCOL_TRACE_LEVEL=0xFF
inside libnfc_nci.conf

One question, did you use --enable-debug at ./configuration? If not, can you also try this?

mwalle, 2022-08-24:
Hi,

> Yes, as soon as we will prepare AN we will also publish patch to GitHub
Ok, great to hear!

> This mean, that with latest code and patch provided in this thread the issue with 77 -> 70 is fixed?
Yes.

> I would ask you to try few things regarding issue that you mention:
> 
> 1. In libnci_nxp.conf please put all loglevels to 0x03 and share logs (in libnfc_nxp.conf)
> 2. Put APPL_TRACE_LEVEL and PROTOCOL_TRACE_LEVEL inside libnfc_nci.conf to 0xFF
> 3. Add --enable-debug to ./configuration step

Please note that compiling with debug symbols (`--enable-debug`) will change the behavior. Now you have to read the card multiple times until the error occurs.

See [output.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9414368/output.txt) for a run with all loglevels enabled. Please note that a short time after the last line, the program will abort with "Segmentation fault (core dumped)"

-michael

skupekA, 2022-09-04:
Hi,

can you please tell me which driver you are using?
Regards,
Andraz

mwalle, 2022-09-05:
Hi,

I'm using the nxpnfc driver from https://github.com/NXPNFCLinux/nxpnfc.

And this is how I compiled the library:
```
# uname -a                                          
Linux host 5.15.28-00011-g821e4429f548-dirty #110 SMP Wed Aug 10 14:04:39 CEST 2|
022 aarch64 GNU/Linux
# cat /etc/debian_version 
10.12
# sh bootstrap
# ./configure --enable-debug
# make
# make install
```

kernel DT fragment:
```
       nxpnfc: nxpnfc@28 {
               compatible = "nxp,nxpnfc";
               reg = <0x28>;
               nxp,nxpnfc-irq = <&gpio5 11 0>;
               nxp,nxpnfc-ven = <&gpio5 12 0>;
               nxp,nxpnfc-fw-dwnld = <&gpio5 10 0>;
       };
```

-michael

skupekA, 2022-09-05:
Can you use gdb and in src/nxp_nci_hal_libnfc-nci/src/include/gki_target.h add "#define GKI_DEBUG TRUE".
They please share logs. We can then find out from where error is coming.
Andraz

mwalle, 2022-09-05:
Hi,

compiling with GKI_DEBUG TRUE leads to the following errors:

```
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -DCONFIG_PATH=\"/usr/local/etc/\" -I./src/include -I./src/nxp_nci_hal_libnfc-nci/src/include -I./src/nxp_nci_hal_libnfc-nci/src/ -I./src/nxp_nci_hal_libnfc-nci/src/gki/ulinux -I./src/nxp_nci_hal_libnfc-nci/src/gki/common -I./src/nxp_nci_hal_libnfc-nci/src/nfa/include -I./src/nxp_nci_hal_libnfc-nci/src/nfa/int -I./src/nxp_nci_hal_libnfc-nci/src/nfc/include -I./src/nxp_nci_hal_libnfc-nci/src/nfc/int -I./src/nxp_nci_hal_libnfc-nci/src/hal/include -I./src/nxp_nci_hal_libnfc-nci/src/hal/int -I./src/nxp_nci_hal_libnfc-nci/src/fuzzers/inc/android/hardware/nfc/1.1/ -I./src/android/utility/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/utils/ -I./src/libnfc-utils/inc -I./src/nfcandroid_nfc_hidlimpl/halimpl/common/ -I./src/android/utility/hardware/ -I./src/android/utility/android/ -I./src/nxp_nci_hal_libnfc-nci/src/gki/ulinux -I./src/nxp_nci_hal_libnfc-nci/src/include -I./src/nfcandroid_nfc_hidlimpl/halimpl/utils/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/inc/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/common/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/dnld/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/hal/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/log/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/tml -I./src/nxp_nci_hal_libnfc-nci/src/fuzzers/inc/android/hardware -I./src/nfcandroid_nfc_hidlimpl/halimpl/self-test -I./src/nfcandroid_nfc_hidlimpl/halimpl/mifare -I./src/nfcandroid_nfc_hidlimpl/halimpl/tml/transport -I./src/nfcandroid_nfc_hidlimpl/halimpl/src/include -I./src/nxp_nci_hal_libnfc-nci/src/fuzzers/inc/ -I./src/nxp_nci_hal_libnfc-nci/utils/include -I./src/nxp_nci_hal_libnfc-nci/utils -I./src/nfcandroid_nfc_hidlimpl/extns/impl/ -I./src/android/secure_element/extns/impl/ -I./src/android/utility/base/ -I./src/libnfc-utils/inc -I./src/service/interface -I./src/service/utils -I./src/nxp_nci_hal_nfc/nci/jni/extns/pn54x/inc -I./src/nxp_nci_hal_nfc/nci/jni/extns/pn54x/src/common -I./src/nxp_nci_hal_nfc/nci/jni/extns/pn54x/src/mifare -I./src/nxp_nci_hal_nfc/nci/jni/ -I./src/nfcandroid_nfc_hidlimpl/halimpl/log -DDEBUG -pthread -w -DNFC_NXP_NOT_OPEN_INCLUDED=TRUE -DNXP_HW_SELF_TEST -DNXP_NFC_NATIVE_ENABLE_HCE=TRUE -DNFC_NXP_LLCP_SECURED_P2P=FALSE -DPH_NCI_NXP_HAL_ENABLE_FW_DOWNLOAD=FALSE -DNXP_NEW_MW_API_COMPATIBILITY -DNXP_EXTNS=TRUE -DLINUX -DSNEP_ENABLED -I./firmware/pn7160 -g -O0 -MT src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.lo -MD -MP -MF src/nxp_nci_hal_libnfc-nci/src/gki/common/.deps/gki_debug.Tpo -c src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc  -fPIC -DPIC -o src/nxp_nci_hal_libnfc-nci/src/gki/common/.libs/gki_debug.o
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc: In function ‘void GKI_PrintBufferUsage(UINT8*, UINT16*)’:
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:75:9: error: ‘memcpy’ was not declared in this scope
         memcpy(p_cur_used, cur, num*2);
         ^~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:75:9: note: ‘memcpy’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:19:1:
+#include <cstring>
 
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:75:9:
         memcpy(p_cur_used, cur, num*2);
         ^~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc: At global scope:
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:167:34: error: variable or field ‘gki_print_buffer_statistics’ declared void
 void gki_print_buffer_statistics(FP_PRINT print, INT16 pool)
                                  ^~~~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:167:34: error: ‘FP_PRINT’ was not declared in this scope
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:167:34: note: suggested alternative: ‘P_PGID’
 void gki_print_buffer_statistics(FP_PRINT print, INT16 pool)
                                  ^~~~~~~~
                                  P_PGID
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:167:56: error: expected primary-expression before ‘pool’
 void gki_print_buffer_statistics(FP_PRINT print, INT16 pool)
                                                        ^~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:207:35: error: variable or field ‘gki_print_used_bufs’ declared void
 GKI_API void gki_print_used_bufs (FP_PRINT print, UINT8 pool_id)
                                   ^~~~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:207:35: error: ‘FP_PRINT’ was not declared in this scope
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:207:35: note: suggested alternative: ‘P_PGID’
 GKI_API void gki_print_used_bufs (FP_PRINT print, UINT8 pool_id)
                                   ^~~~~~~~
                                   P_PGID
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:207:57: error: expected primary-expression before ‘pool_id’
 GKI_API void gki_print_used_bufs (FP_PRINT print, UINT8 pool_id)
                                                         ^~~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:257:22: error: variable or field ‘gki_print_task’ declared void
 void gki_print_task (FP_PRINT print)
                      ^~~~~~~~
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:257:22: error: ‘FP_PRINT’ was not declared in this scope
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:257:22: note: suggested alternative: ‘P_PGID’
 void gki_print_task (FP_PRINT print)
[... cut ...]
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:345:34: error: ‘print’ cannot be used as a function
             print("%08lX, ", s[i]);
                                  ^
src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.cc:349:15: error: ‘print’ cannot be used as a function
     print("\n");
               ^
make[1]: *** [Makefile:1894: src/nxp_nci_hal_libnfc-nci/src/gki/common/gki_debug.lo] Error 1
make[1]: Leaving directory '/root/linux_libnfc-nci'
make: *** [Makefile:893: all] Error 2
```

-michael

skupekA, 2022-09-05:
What about only with gdb? -> what is output there?

mwalle, 2022-09-05:
```
(gdb) run poll
Starting program: /usr/local/sbin/nfcDemoApp poll
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
################################################################################                                               #########
##                                       NFC demo                                                                                     ##
################################################################################                                               #########
##                                 Poll mode activated                                                                                ##
################################################################################                                               #########
[New Thread 0xfffff76871c0 (LWP 7719)]
                              ... press enter to quit ...

[New Thread 0xfffff6e861c0 (LWP 7720)]
[New Thread 0xfffff66851c0 (LWP 7721)]
[New Thread 0xfffff5e841c0 (LWP 7722)]
[Thread 0xfffff66851c0 (LWP 7721) exited]
[New Thread 0xfffff5682900 (LWP 7723)]
[New Thread 0xfffff66851c0 (LWP 7724)]
[New Thread 0xfffff56601c0 (LWP 7725)]
[New Thread 0xfffff4e5f1c0 (LWP 7726)]
[New Thread 0xffffdffff1c0 (LWP 7727)]
[Thread 0xffffdffff1c0 (LWP 7727) exited]
Waiting for a Tag/Device...

        NFC Tag Found

        Type :         'Type A - Mifare Ul'
        NFCID1 :        '04 EF C6 AA 28 63 80 '
[New Thread 0xffffdffff1c0 (LWP 7728)]
                Record Found :
                                NDEF Content Max size :     '137 bytes'
                                NDEF Actual Content size :     '11 bytes'
                                ReadOnly :                      'FALSE'
                                Type :                 'Text'
                                Lang :                 'en'
                                Text :                 'test'

                11 bytes of NDEF data received :
                D1 01 07 54 02 65 6E 74 65 73 74

        NFC Tag Lost

Waiting for a Tag/Device...

[Thread 0xffffdffff1c0 (LWP 7728) exited]
        NFC Tag Found

        Type :         'Type A - Mifare Ul'
        NFCID1 :        '04 EF C6 AA 28 63 80 '
[New Thread 0xffffdffff1c0 (LWP 7729)]
                Record Found :
                                NDEF Content Max size :     '137 bytes'
                                NDEF Actual Content size :     '11 bytes'
                                ReadOnly :                      'FALSE'

Thread 1 "nfcDemoApp" received signal SIGSEGV, Segmentation fault.
PrintNDEFContent (TagInfo=0xfffffffff678, NDEFinfo=0xfffffffff9e0,
    ndefRaw=0x0, ndefRawLen=11) at demoapp/main.c:816
816                             TextContent[res] = '\0'; /* Add extra character                                                for proper display */
(gdb) bt
#0  PrintNDEFContent (TagInfo=0xfffffffff678, NDEFinfo=0xfffffffff9e0,
    ndefRaw=0x0, ndefRawLen=11) at demoapp/main.c:816
#1  0x0000aaaaaaaa4358 in WaitDeviceArrival (mode=1, msgToSend=0x0, len=0)
    at demoapp/main.c:1285
#2  0x0000aaaaaaaa590c in cmd_poll (arg_len=0, arg=0xfffffffffbb8)
    at demoapp/main.c:1886
#3  0x0000aaaaaaaa6088 in main (argc=2, argv=0xfffffffffba8)
    at demoapp/main.c:2145
(gdb)
```

skupekA, 2022-09-05:
Okay, now we can reproduce an issue. Will debug and then came back.