NXPNFCLinux/linux_libnfc-nci #144: Error during config reaout on Linux 64bit arm platform

swaechterSX, 2022-07-11:
When using branch [NCI2.0_PN7160](https://github.com/NXPNFCLinux/linux_libnfc-nci/compare/NCI2.0_PN7160)(commit ee8ad71123b07b5a33012df0c614ab3e9278c080) on an imx8mm linux release readout of both
NFA_PROPRIETARY_CFG and POLLING_TECH_MASK and P2P_LISTEN_TECH_MASK form libnfc-nci.conf failed.

The same release on 32bit platfrom (Raspberry OS) worked fine.

This is due to bugs in src/android/utility/base/strings.cpp:android::base::Split and src/nxp_nci_hal_libnfc-nci/utils/config.cc:GetNumValue.

Patches:
[https://gist.github.com/swaechterSX/d894040904cfe63a815f35fea9cb366c](url)
[https://gist.github.com/swaechterSX/c8d85efc8ccf7669147d88d13d2836b6](url)

mwalle, 2022-07-14:
Thanks! This also fixes #143. I'm curious. Have you been able to successfully read a non-empty tag? For me, the demo app will do a coredump, which makes me wonder if this is also a 64bit bug.

swaechterSX, 2022-07-14:
Good to know. 

No I haven’t made much more tests with it (as I wasted quite a bit of time with it already and thought for the longest time our hardware was buggy).
I did inform NXP and asked them if they made tests with this lib on an linux  (not android) 64bit platform as I already feared that there might be more issues like the ones I found.
Next week I will make some further tests with the nfc reader and let you know if it also crashes for me with empty tags.

skupekA, 2022-07-28:
Hi @swaechterSX,

is it possible to send us logs?

Regards,
Andraz

swaechterSX, 2022-07-28:
Here the logs its same sw ones on rasparry (running 32bit rasparry os) and once an imx8mm board running 64bit release
base on imx-yocto.
[imx8mm-pn7160.log](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9207424/imx8mm-pn7160.log)
[rasparry-good.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9207425/rasparry-good.txt)

The relevant line for this issue is “nfcManager_enableDiscovery: enter; tech_mask = “.
The problem is that techmask was not set correcty (Even so it was set in the conf file). 
As described in the issue I tracked this to a failure during config readout and 0xff is the default value).

Further not that with commit (ee8ad71123b07b5a33012df0c614ab3e9278c080) the library simply crashed during readout of NFA_PROPRIETARY_CFG in libnfc-nci.conf on the imx8mm-mini. 
This was due to the error in andriod::base::Split. 
So this log is of commit ee8ad71123b07b5a33012df0c614ab3e9278c080 + [0001-Fix-andriod-base-Split-for-64bit-arm-architecture.patch](https://gist.github.com/swaechterSX/118019e785013640f154d6fb78925c5e#file-0001-fix-andriod-base-split-for-64bit-arm-architecture-patch)

skupekA, 2022-07-28:
@swaechterSX can you also share config files?

swaechterSX, 2022-07-28:
They are exactly the same than your config files in ee8ad71123b07b5a33012df0c614ab3e9278c080 but here you go

[libnfc-nci.conf.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9211897/libnfc-nci.conf.txt)
[libnfc-nxp.conf.txt](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9211898/libnfc-nxp.conf.txt)
(I had to rename them to txt for git)

I only switched the NXP_TRANSPORT to ALT_I2C (on the imx8mm I used both userspace and nxpnfc driver, but this is unrelated to the issue).

skupekA, 2022-08-22:
@swaechterSX, We check and provide this patch
[linux_64bitOs.zip](https://github.com/NXPNFCLinux/linux_libnfc-nci/files/9393945/linux_64bitOs.zip)

Is there any other issue?

Regards,
Andraz

StephanWaechter, 2022-08-23:
This is exactly the patch I provided 40 days ago (https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/144#issue-1300677788)? Yes this fixes the issue of the read out (as was allready confirmed by me and @mwalle). The question is are there more issues like this in the library? Did you test it on a 64bit linux (without android)? 
Did you check the issue @mwalle noticed in https://github.com/NXPNFCLinux/linux_libnfc-nci/issues/144#issuecomment-1184307142. Are you planning to add this to branch [NCI2.0_PN7160](https://github.com/NXPNFCLinux/linux_libnfc-nci/compare/NCI2.0_PN7160).
I currently don't work on embedded linux so I will not continue following up one this issue.
But everyone who wants to use this library on a 64bit platform linux (like the many sold by nxp) will stumple over this issue so I would expect it to be fixed in [NCI2.0_PN7160](https://github.com/NXPNFCLinux/linux_libnfc-nci/compare/NCI2.0_PN7160).

skupekA, 2022-08-23:
Hi,
yes, this was tested on 64bit Linux. Also we checked issue that @mwalle mention and noticed that the configuration was wrong (still waiting for response).
We are working on updated application note and when it is ready we will update code as well.
Regards